OpenKinect/libfreenect2 #795: Speeding up computation at the driver

DepSys4SmartHomes, 2017-02-14:
Libfreenect2 is pretty fast when it is run on an Intel Core i5 or a Intel Core i7 processor. However, when an ARM processor is used (e.g., in an Odroid), the driver produces frames at 13 FPS, instead of 30 FPS as in Kinect specification. I see a fair amount of computation happening in the driver. I will appreciate if anyone has any suggestions to speed up the computation at the driver for a low power processor. Thank you!

floe, 2017-02-14:
I know that people have been using OpenCL for acceleration on some ARM boards, e.g., the Odroid XU4. But I don't think you can get a lot of speedup from the CPU decoder, except if you rewrite it in NEON assembly.
-- 
SENT FROM MY PDP-11

xlz, 2017-02-20:
There is little room for optimization in the algos. There is some room for optimization in data processing. 

Odroid etc don't/shouldn't use the ARM processor for the data, they use graphics coprocessor with OpenGL 3.1+ capacities.

closed 2017-02-20 by xlz