OpenKinect/libfreenect2 #798: cuda_depth_packet_processor.cu(31) helper_math.h file is not found in windows 8.1 using VS2013

RajaMOeez, 2017-02-20:
I have all the things configured according to the windows configuration but when Im trying to build the this error is occur could not found the header file **helper_math.h** I have also tried to set the path manually but it not works. is anyone know about this issue please help me or guide me. I'll be very grateful him/her  
[image]

xlz, 2017-02-20:
I honestly can't reproduce this. `helper_math.h` is right here
```
C:\Users\stars\Documents\GitHub\libfreenect2\build [master]> ls $env:NVCUDASAMPLES8_0_ROOT/common/inc/helper_math.h

    Directory: C:\ProgramData\NVIDIA Corporation\CUDA Samples\v8.0\common\inc

Mode                LastWriteTime     Length Name
----                -------------     ------ ----
-a---        10/28/2015   5:07 AM      38763 helper_math.h
```

RajaMOeez, 2017-02-20:
thanks for the quick response @xlz  Basically Im newby here with kinect and facing difficulties to tackle it. Can't understand it, if you know how we can solve it in **windows** with this error. that I'll be very thankful to you @xlz .
Best Regard 
MOeez

xlz, 2017-02-20:
Provide the full log of commands you've run starting from downloading the source code `git clone`.

RajaMOeez, 2017-02-20:
please give me a minute I'll provide you.

RajaMOeez, 2017-02-20:
[image]

RajaMOeez, 2017-02-20:
[image]

RajaMOeez, 2017-02-20:
[image]

xlz, 2017-02-20:
The log is incomplete. I can't find anything.

RajaMOeez, 2017-02-20:
**sorry I have update it please see it again**

RajaMOeez, 2017-02-20:
Mr @xlz  I analogize for this. Will you please do me another favor please come-on skype or any other remote access so maybe it will more clear to you. Its my humble request if you can come.

xlz, 2017-02-20:
Here is the log of a successful build. Check for yourself.
```
C:\Users\stars\Documents\GitHub\libfreenect2\build [master]> cmake .. -G "Visual Studio 12 2013 Win64"
-- The C compiler identification is MSVC 18.0.40629.0
-- The CXX compiler identification is MSVC 18.0.40629.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Performing Test LIBUSB_WITH_LIBUSBK
-- Performing Test LIBUSB_WITH_LIBUSBK - Success
-- Performing Test LIBUSB_WITH_USBDK
-- Performing Test LIBUSB_WITH_USBDK - Success
-- Found LibUSB: C:/Users/stars/Documents/GitHub/libfreenect2/depends/libusb/MS64/dll/libusb-usbdk-1.0.lib  
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: C:/libjpeg-turbo64/lib/turbojpeg.lib  
-- Found GLFW3: C:/Users/stars/Documents/GitHub/libfreenect2/depends/glfw/lib-vc2013/glfw3dll.lib  
-- Found OpenGL: opengl32  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0/lib/x64/OpenCL.lib (found version "1.2") 
-- Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0 (found version "8.0") 
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Linking with these libraries: 
 C:/Users/stars/Documents/GitHub/libfreenect2/depends/libusb/MS64/dll/libusb-usbdk-1.0.lib
 C:/libjpeg-turbo64/lib/turbojpeg.lib
 C:/Users/stars/Documents/GitHub/libfreenect2/depends/glfw/lib-vc2013/glfw3dll.lib
 opengl32
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0/lib/x64/OpenCL.lib
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v8.0/lib/x64/cudart_static.lib
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Using in-tree freenect2 target
-- Found OpenNI2: C:/Program Files/OpenNI2/Lib  
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/stars/Documents/GitHub/libfreenect2/build

C:\Users\stars\Documents\GitHub\libfreenect2\build [master]> cmake --build . --config RelWithDebInfo --target install
Microsoft (R) Build Engine version 12.0.40629.0
[Microsoft .NET Framework, version 4.0.30319.42000]
[... 409 lines cut ...]
  C:\Users\stars\Documents\GitHub\libfreenect2\include\internal\libfreenect2/protocol/response.h(324): warning C4200: nonstandard extension used : zero-sized array in struct/union [C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj]
  C:\Users\stars\Documents\GitHub\libfreenect2\include\internal\libfreenect2/protocol/response.h(324): warning C4200: nonstandard extension used : zero-sized array in struct/union [C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj]

"C:\Users\stars\Documents\GitHub\libfreenect2\build\install.vcxproj" (default target) (1) ->
"C:\Users\stars\Documents\GitHub\libfreenect2\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Users\stars\Documents\GitHub\libfreenect2\build\examples\Protonect.vcxproj" (default target) (4) ->
"C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj" (default target) (5) ->
(Link target) -> 
  cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj : warning LNK4217: locally defined symbol ??0Frame@libfreenect2@@QEAA@_K00PEAE@Z (public: __cdecl libfreenect2::Frame::Frame(unsigned __int64,unsigned __int64,unsigned __int64,unsigned char *)) imported in function "public: __cdecl libfreenect2::CudaFrame::CudaFrame(class libfreenect2::Buffer *)" (??0CudaFrame@libfreenect2@@QEAA@PEAVBuffer@1@@Z) [C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj]
  cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.obj : warning LNK4049: locally defined symbol ??0Frame@libfreenect2@@QEAA@_K00PEAE@Z (public: __cdecl libfreenect2::Frame::Frame(unsigned __int64,unsigned __int64,unsigned __int64,unsigned char *)) imported [C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj]
[... cut ...] freenect2\build\freenect2.vcxproj]
  cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.obj : warning LNK4049: locally defined symbol ?getGlobalLogger@libfreenect2@@YAPEAVLogger@1@XZ (class libfreenect2::Logger * __cdecl libfreenect2::getGlobalLogger(void)) imported [C:\Users\stars\Documents\GitHub\libfreenect2\build\freenect2.vcxproj]

    41 Warning(s)
    0 Error(s)

Time Elapsed 00:00:23.41
```
> Will you please do me another favor please come-on skype or any other remote access

No

RajaMOeez, 2017-02-20:
The C compiler identification is MSVC 18.0.40629.0
The CXX compiler identification is MSVC 18.0.40629.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
using tinythread as threading library
Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
Performing Test LIBUSB_WITH_LIBUSBK
Performing Test LIBUSB_WITH_LIBUSBK - Success
Performing Test LIBUSB_WITH_USBDK
Performing Test LIBUSB_WITH_USBDK - Success
CMake Error at C:/Program Files/CMake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES LibUSB_INCLUDE_DIRS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:102 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:93 (FIND_PACKAGE)

Configuring incomplete, errors occurred!
See also "C:/Users/MOeez Raja/Documents/Visual Studio 2013/libfreenect2/build/CMakeFiles/CMakeOutput.log".

RajaMOeez, 2017-02-20:
@xlz  Sir I analogize for nonsense thing I have asked you.
Basically I was building **LIBFREENECT2** with **Windows Git Shell** that's why maybe you didn't understand my problem and also I can't make you understand my problem clearly but now its a complete thing that I have tried to build it with **GUI CMAKE** please one more time see it.
using tinythread as threading library
Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
Linking with these libraries: 
 C:/Users/MOeez Raja/Documents/Visual Studio 2013/Projects/libfreenect2-master/depends/libusb/MS64/dll/libusb-usbdk-1.0.lib
 C:/libjpeg-turbo64/lib/turbojpeg.lib
 C:/glfw/lib-vc2013/glfw3dll.lib
 opengl32
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v7.5/lib/x64/OpenCL.lib
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v7.5/lib/x64/cudart_static.lib
Configurating examples
Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
Using in-tree freenect2 target
Feature list:
  CUDA    yes
  CXX11    disabled
  Examples    yes
  OpenCL    yes
  OpenGL    yes
  OpenNI2    yes
  TegraJPEG    no
  Threading    tinythread
  TurboJPEG    yes
  VAAPI    no
  VideoToolbox    no (Apple only)
**Configuring done
Generating done**

**ERROR**

1. **Error**		error C1083: Cannot open include file: 'helper_math.h': No such file or directory	C:\Users\MOeez Raja\Documents\Visual Studio 2013\Projects\libfreenect2-master\src\cuda_depth_packet_processor.cu	31	1	freenect2
2. **Error**		error LNK1104: cannot open file '..\lib\Debug\freenect2d.lib'	C:\Users\MOeez Raja\Documents\Visual Studio 2013\Projects\libfreenect2-master\build\examples\LINK	Protonect
3.**Error**	error LNK1104: cannot open file 'lib\Debug\freenect2d.lib'	C:\Users\MOeez Raja\Documents\Visual Studio 2013\Projects\libfreenect2-master\build\LINK	freenect2-openni2

xlz, 2017-03-29:
Can't reproduce.

closed 2017-03-29 by xlz

AAcquier, 2018-04-25:
Did it ever got solved? I got the same issue with cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj

AAcquier, 2018-04-25:
Never mind, it was due to a mismatch of the version of VS used in the command line and the one installed on the PC

SizheAn, 2022-02-16:
> Never mind, it was due to a mismatch of the version of VS used in the command line and the one installed on the PC

How did you solve it? Did you just install a lower version of VS?

AAcquier, 2022-02-16:
Hi SizheAn, Yes as I recall it, I removed the VS version that I got to install the one that suited libfreenect.