OpenKinect/libfreenect2 #802: when I install libfreenect2 until "./bin/Protonect",something wrong

UThinkD, 2017-03-09:
# Version, Platform, and Hardware Bug Found:
Ubuntu14.04  ROS Indigo  kinect v2,   i7-6700hq  16G RAM

wang@vulcan:~/catkin_ws/libfreenect2/build$ ./bin/Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:8 with serial 004032562447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
wang@vulcan:~/catkin_ws/libfreenect2/build$ 

Seveal days ago, I have installed it sucessfully. But today ,when I try to connect with the kinect2 again ,something goes wrong. Then I have tried reinstall from the Ununtu,however, it doesn't work,please help me?

floe, 2017-03-10:
Well, if it worked before, then you probably changed something in the meantime - did you update your system, for example? New kernel? Look in the README for how to enable USB debugging, and run again.

vacky11, 2017-03-20:
@UThinkD 
I am facing the exact same problem. It was working fine for me for both the KinectV2 I had. But now one KinectV2 is working but for other one, I am getting the same error. Did you find any solution for this?

floe, 2017-03-20:
Let me reiterate: if it worked before, then you probably changed something in the meantime - did you update your system, for example? New kernel? Look in the README for how to enable USB debugging, and run again.

UThinkD, 2017-03-25:
@floe  Firstly thank you for your help. Probably the problem comes from the system update. When  I install the Ubuntu system ,I chose the default installation rather than manually set the partition. So , I re-install the system and set the "swap" double the physical memory. Now problem has solved.

UThinkD, 2017-03-25:
@vacky11  I think the problem is insufficient virtual memory. It is recommended be set as double your physical memory. I hope this can be helpful to you!

xlz, 2017-03-28:
User reports problem was solved.

closed 2017-03-28 by xlz