OpenKinect/libfreenect2 #804: undefined reference to `clRetainContext@OPENCL_1.0'

dingh, 2017-03-11:
I recently build libfeenect2 and I was trying to make some changes in the Protonect example. While compiling, I was getting the following error:

Scanning dependencies of target Protonect
[ 50%] Building CXX object CMakeFiles/Protonect.dir/Protonect.cpp.o
[100%] Linking CXX executable Protonect
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainDevice@OPENCL_1.2'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetExtensionFunctionAddress@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clBuildProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseDevice@OPENCL_1.2'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformIDs@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clSetKernelArg@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetPlatformInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseProgram@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseEvent@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueReadBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clWaitForEvents@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainCommandQueue@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueWriteBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetProgramBuildInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clRetainMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseContext@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueUnmapMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clGetDeviceInfo@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueNDRangeKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateProgramWithSource@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseKernel@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clReleaseMemObject@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clEnqueueMapBuffer@OPENCL_1.0'
/usr/local/lib/libfreenect2.so: undefined reference to `clCreateCommandQueue@OPENCL_1.0'
collect2: error: ld returned 1 exit status
CMakeFiles/Protonect.dir/build.make:95: recipe for target 'Protonect' failed
make[2]: *** [Protonect] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Protonect.dir/all' failed
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I am running it on ubuntu 16.04, OpenCL 1.2 Cuda 8.0. 
I suspect that the problem is caused by two libOpenCL.so files on my machine (one in /usr/x86/lib/ the other under CUDA), however I don't know how to fix it.

Any help will be appreciated.

-Hao

dingh, 2017-03-11:
I was able to compile the example after recompile the libfreennect2 with OpenCL option turned off. 
However I would still like to know how to make it work with OpenCL

xlz, 2017-03-14:
OK, so this only happens with third party compilation using installed libfreenect2.so, which is why I didn't see it happen with Protonect. To reproduce, first `sudo make install` libfreenect2, then `cd examples; mkdir build && cd build; cmake ..; make`.

The cause of the issue is this:

First, libfreenect2 is built with ocl-icd which provides versioned libOpenCL.so. Therefore libfreenect2.so requires versioned symbols.
Then, when compiling an app with this libfreenect2, CUDA's libOpenCL.so will take precendence somehow and CUDA's libOpenCL.so is unversioned. So GCC would enforce versioning and report undefined references to `symbol@version`. But at runtime this versioning doesn't seem to be strictly enforced.

Still thinking about what to do with it.

xlz, 2017-03-14:
OK. I think this is a fuckup by Nvidia showing obvious lack of care about compatibility. The problem is not limited to us. Any library that is built with ocl-icd-libopencl1 (libarrayfire-opencl3, libasl0, libclblas2, libclfft2, etc.) will fail to be linked against with this setup of CUDA even without `LD_LIBRARY_PATH`.

The package `cuda-cudart-8-0` installs `cuda-8-0.conf` into `/etc/ld.so.conf.d` which puts `/usr/local/cuda-8.0/targets/x86_64-linux/lib` above `/usr/lib/x86_64-linux-gnu` during compile time and run time. libfreenect2 happens to find and use `/usr/lib/x86_64-linux-gnu/libOpenCL.so` by full path so the problem only reveals itself at later run time and thiry-party compile time.

The libOpenCL.so provided by CUDA is mostly a worse copy of libOpenCL.so of ocl-icd-libopencl1 without versioned symbols. And it takes precedence.

Solutions:

The solution to the root cause is `sudo mv /etc/ld.so.conf.d/cuda-8-0.conf /etc/ld.so.conf.d/zz_cuda-8-0.conf` but this needs to be done every time the package `cuda-cudart-8-0` is updated. Nvidia must take action if this change is to be persistent.

If you have a prebuilt library linked with ocl-icd (e.g. libfreenect2) and you can't rebuild it, use `LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu make` to build your own program to override the search path in `cuda-8-0.conf`.

If you can rebuild the library requiring ocl-icd (e.g. libfreenect2), built it with `cmake .. -DCMAKE_INSTALL_RPATH=/usr/lib/x86_64-linux-gnu` to override the search path in `cuda-8-0.conf`. But this is an ugly hack which you should stop using after Nvidia properly resolves this problem.

closed 2017-03-14 by xlz

xlz, 2017-03-14:
I added this workaround so you don't need to do anything now.

DufeRob, 2017-03-14:
I recompile libfreenect2 with the new cmake. Protonect works (it worked before), but I still can't compile ROS catkin package iai with catkin_make. 
freenect2/lib/libfreenect2.so: undefined reference to `clReleaseMemObject@OPENCL_1.0' and so on
How can I avoid this error?

xlz, 2017-03-14:
The new libfreenect2.so should have a RPATH. `objdump -x libfreenect2.so | grep RPATH`.

DufeRob, 2017-03-14:
I have nothing with that command. What should I do?

xlz, 2017-03-14:
Clean up libfreenect2 build directory. Redo `cmake ..; make; sudo make install` for libfreenect2. And try catkin again.

svenevs, 2017-06-08:
@xlz by `/usr/lib/x86_64-linux-gnu` you mean more generally wherever `libOpenCL.so.1.0` resides, yes?  E.g.,

```console
$ locate *OpenCL*.so
/usr/lib/libOpenCL.so
/usr/lib64/libOpenCL.so
/usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so
```

I tried both of these non-cuda paths, but I still can't seem to get things to check out.  It's not really a big deal, I'm not using OpenCL so I just disabled it.  I ask because my `libusb-1.0` requires an RPATH to be set as well, and thought maybe this could be causing issues in the build system.  If I set the RPATH to either have both `/usr/lib64` and my custom `libusb`, or even just `/usr/lib64` (which should break the `libusb` dep?), it'll always give the above mentioned link error for external projects (thanks nvidia........).  Anyway, I gave [the wiki suggestions](https://cmake.org/Wiki/CMake_RPATH_handling) a whirl but they all result in the same issue when compiling external projects.

commit 6304ac64 (2017-03-14): cmake: Work around CUDA 8 overriding OpenCL path

Fix #804.