OpenKinect/libfreenect2 #806: Help to get driver working with OpenNI2 (Linux)

Rep8, 2017-03-12:
I've got the libfreenect2 driver with multiple kinect v2's with the protonect viewing tool, however I cannot work out how to get OpenNI2 up and running.

I only get the device not found prompt when trying anything.
I compiled the libfreenect2 driver for openni2 (pretty sure I did it correctly), then copied the driver files to multiple openni2/drivers locations just to be sure.

Any help/suggestions would be appreciated.

xlz, 2017-03-13:
What is your question?

Rep8, 2017-03-13:
Sorry I was pretty broad.

I want to be able to get the NiViewer2 for openni2 running with kinect2. Then from that NITE working along it.

I've got the protonect viewer working but the NiViewer2 just returns unable to open device when it tries to load the drivers.

xlz, 2017-03-13:
Log?

Rep8, 2017-03-13:
[image]

xlz, 2017-03-13:
Did you install libfreenect2.so.0.2? Where is it?

Rep8, 2017-03-13:
Okay I got it working, completely misread the log. I have added those drivers to the specified folder.
Only issue is in the NiViewer2 program, I have the depth and color images but not the ir camera image, is this normal?

Thank you very much though.

xlz, 2017-03-14:
I think IR channel is available but just not shown?

closed 2017-03-14 by xlz