OpenKinect/libfreenect2 #810: Works, but not via

Masterxilo, 2017-03-16:
1. I installed the UsbDk driver (I want the SDK driver to keep working without manual switching).
2. I downloaded libfreenect2-0.2.0-usbdk-vs2015-x64.zip
3. I copied the contents of "libfreenect2-0.2.0-usbdk-vs2015-x64\bin\" to "OpenNI2\Tools\OpenNI2\Drivers" of an OpenNI 2.2.0.33 installation
4. Running the newly copied "OpenNI2\Tools\OpenNI2\Drivers\Protonect.exe" works (shows all recorded images) and produces to following in its console
```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\...
pu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @27:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 174127640347
device firmware: 4.0.3912.0
```
5. Running "OpenNI2\Tools\NiViewer.exe" produces 
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @27:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found 0 devices
[Error] [Freenect2Impl] requested device 0 is not connected!
```
and crashes. Obviously, it unnecessarily tries to scan the devices a second time and then fails to find it when run from NiViewer?
6. When I now copy "OpenNI2\Tools\OpenNI2" to "OpenNI2\Samples\Bin\OpenNI2" and run "OpenNI2\Samples\Bin\SimpleViewer.exe" I get almost the same:
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @27:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found 0 devices
After initialization:

[Error] [Freenect2Impl] requested device 0 is not connected!
```

What could I try? The libusbK driver? (I'd rather not)

I have no issues with OpenNI2 and a Kinect v1 (Xbox 360) or an ASUS Xtion Pro Live sensor.

Windows 8.1, x64

xlz, 2017-03-17:
Can you try usbdk 1.0.16?

I tried a few things. The 0.2.0 binary release works with usbdk 1.0.16 but not 1.0.17. When I rebuild libusb it seems to work with usbdk 1.0.17 too.

Masterxilo, 2017-03-18:
Thanks. NiViewer works now with usbdk 1.0.16, it shows
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @9:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @9:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
```

However, the SimpleViewer does not open up properly, stating
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @9:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @9:2 with serial 174127640347
[Info] [Freenect2Impl] found 1 devices
After initialization:

[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
Error - expect color and depth to be in same resolution: D: 640x480, C: 1920x1080
```
and then it only shows the console, no window. But maybe this is an error in SimpleViewer, maybe it does not properly communicate to the device that it wants both streams in the same resolution.

xlz, 2017-03-18:
SimpleViewer is simple. It simply requires the same resolution while NiViewer does not.

closed 2017-03-18 by xlz