OpenKinect/libfreenect2 #811: vaDisplayIsValid random Segmentation fault

alexge233, 2017-03-16:
Overview Description:

I am getting random SEGFAULT from `vaDisplayIsValid`.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`: cec93c7 Merge pull request #765 from xlz/disable-default-constructors
2. Ubuntu 16.04

Steps to Reproduce:

1. I am using the default constructor as shown in the examples, encapsulated in my own class.
2. At random intervals, I get a segmentation fault

```cpp
    if (handle.enumerateDevices() == 0) {
       throw std::runtime_error("no kinect v2 connected!");
    }
    std::string serial = handle.getDefaultDeviceSerialNumber();
    pipeline = new libfreenect2::CpuPacketPipeline();
    assert(pipeline);
    dev = handle.openDevice(serial, pipeline);
    if (dev == 0) {
        throw std::runtime_error("failed to open kinect v2");
    }
    int types = libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth;
    listen = std::make_unique<libfreenect2::SyncMultiFrameListener>(types);
    dev->setIrAndDepthFrameListener(&*listen);
    dev->setColorFrameListener(&*listen);
    if (!dev->start()) {
        throw std::runtime_error("failed to start kinect streams");
    }
    registration = new libfreenect2::Registration(dev->getIrCameraParams(), 
                                                  dev->getColorCameraParams());
```

I was getting random SEGFAULT, so I kept running it under gdb, and at some stage I found out:

```
[New Thread 0x7fffd178a700 (LWP 23148)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 001680660547
[Info] [Freenect2Impl] found 1 devices

Thread 1 "mario_node" received signal SIGSEGV, Segmentation fault.
0x00007fffeeda3347 in vaDisplayIsValid () from /usr/lib/x86_64-linux-gnu/libva.so.1
(gdb) bt
#0  0x00007fffeeda3347 in vaDisplayIsValid () from /usr/lib/x86_64-linux-gnu/libva.so.1
#1  0x00007ffff576a5c7 in libfreenect2::VaapiRgbPacketProcessorImpl::initializeVaapi() () from /home/mario/freenect2/lib/libfreenect2.so.0.2
#2  0x00007ffff5769408 in libfreenect2::VaapiRgbPacketProcessor::VaapiRgbPacketProcessor() () from /home/mario/freenect2/lib/libfreenect2.so.0.2
#3  0x00007ffff5756ee5 in libfreenect2::CpuPacketPipeline::CpuPacketPipeline() () from /home/mario/freenect2/lib/libfreenect2.so.0.2
#4  0x0000000000413e60 in freenect2::freenect2() ()
```

My system has the apt-get version of libva: 1.7.0-1 installed.
I understand this is quite likely a bug in libva, so I will manually update it and open an issue there, however I'd like to know if this has been experienced before, and if removing libva will significantly slow down libfreenect2.

xlz, 2017-03-16:
Can you add `LOG_INFO << "display=" << display;` above this line https://github.com/OpenKinect/libfreenect2/blob/master/src/vaapi_rgb_packet_processor.cpp#L266
and report what it prints?

xlz, 2017-03-16:
> if this has been experienced before

No

> if removing libva will significantly slow down libfreenect2.

Depends. It will be slower but not always "significantly."

alexge233, 2017-03-16:
@xlz Hi, I'm going to have to try this tomorrow, and will get back to you on it. 
What do you use libva for?  RGB encoding/decoding, PCL or something else?
I'm asking because I'm only interested in PCL and depth, not RGB.

xlz, 2017-03-16:
libva is for RGB. If you don't need RGB you would use this instead
```
int types = libfreenect2::Frame::Ir | libfreenect2::Frame::Depth;
```

Yeah, but a bug is a bug.

xlz, 2017-03-16:
Oh, and you also need to use an environment variable to disable libva `LIBVA_DRIVER_NAME=none` or rebuild libfreenect2 without libva `cmake -DENABLE_VAAPI=OFF`.

alexge233, 2017-03-16:
I'm away from the robot right now, but I can tell you that I'm `ssh`ing into it, but I did export a `display:=0` when using `ssh -Y -X`.

alexge233, 2017-03-17:
Hi @xlz I rebuilt with the added line, but it doesn't print anything (I'm using the default logger). I've also disabled VAAPI whilst doing that. Should I try echoing on stdout?

xlz, 2017-03-17:
Essentially I want to find out what kind of value does display have and where exactly does it crash in libva.

alexge233, 2017-03-17:
If I disable libva in CMAKE then I'm sure that display won't echo, right?

xlz, 2017-03-17:
If disabled at build time it won't print.

alexge233, 2017-03-17:
ok, so I should re-enable and try again? Because right now I've built it without VA

xlz, 2017-03-17:
Yes.

alexge233, 2017-03-17:
Hi, I've added the default logger, rebuilt using libva, added the line, but I still don't see that particular log entry...

xlz, 2017-03-17:
Then it's probably working as intended.

alexge233, 2017-03-22:
I'm closing this as I'm going to assume its a bug in `libva`. 
Sadly the issue opened there hasn't received any responses whatsoever.
I'll be using libfreenect without `libva` from now on.
@xlz Thanks for the help !

closed 2017-03-22 by alexge233