OpenKinect/libfreenect2 #812: Windows 10 USB Connection Failing

wired365, 2017-03-20:
Overview Description:

I'm running on a Windows 10 PC and I have followed the README steps numerous times with no luck. I get everything to install and compile properly, or so it appears. When I go to run the Protonect example I run into the issue of it being able to find the Kinect and Serial but it is unable to request a USB connection. I've traced the issue down to getting LIBUSB_ERROR_OTHER which I think is coming from usbdk throwing an INVALID_HANDLE_VALUE error when it attempts to open the device. From the searching online I can't seem to find anyone else with this issue and I am wondering if anyone knows a solution or if this is a bug?

Version, Platform, and Hardware Bug Found:
- Running Windows 10 on a server machine
- USB 3.0 port is being used.
- Using UsbDk driver
- Visual Studio 2015 Professional
- CMake 3.8.0
- No optional software installed

Error Output:
```
$ ./install/bin/Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: U:\Repositories\libfreenect2\build\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @12:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
libusb_open returned: -99
[Error] [Freenect2Impl] failed to open Kinect v2: @12:2 LIBUSB_ERROR_OTHER Other error
failure opening device!
```

xlz, 2017-03-21:
Does libusbk work?

Does the Official SDK work?

wired365, 2017-03-21:
When I used libusbk it is unable to find that there is any kinect connected to the computer.

The official SDK appears to work. Using the Kinect Studio that comes with the software I was able to get all of the video streams from the camera to work just fine. I haven't tried a command line application or code yet.

xlz, 2017-03-21:
There is very little we can do here if the problem is in libusb. You can try out early versions of https://github.com/libusb/libusb

wired365, 2017-03-22:
I'll give that a try. If I am still having issues should I post it to the libusb or the usbDk github page. I know usbDk has had some issues with Windows 10 and USB ports. Also do you have a recommendation for finding out why the connection is failing? As mentioned in my original post the usbi_backend seems to be returing an INVALID_HANDLE_VALUE, but beyond that I can't figure out why it failed to connect.

xlz, 2017-03-22:
Not so soon. If so I have more for you to try.

wired365, 2017-03-24:
I tried rolling back to version 1.0.20 of libusb this morning but I ran into a problem. It doesn't appear that there is a libusb_usbdk visual studio project for me to build for libfreenect2 to use. Does this mean I have to use the alternative libusbk software or did I miss something?

wired365, 2017-03-24:
So I ended up trying a number of things and ultimately what ended up semi working was keeping everything at the newest version except I rolled back my usbDk version to 1.00-16 instead of 1.00-17. When I run with that, I no longer get invalid USB device handles. I do however run into another error:

```
$ ./install/bin/Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: U:\Repositories\libfreenect2\build\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @8:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
Pipeline Not-Used
libusb_open returned: 0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
```

Is this now a known problem?

xlz, 2017-03-24:
So kind of an issue with usbdk 1.0.17. See #810.

Does the binary release protonect.exe work with usbdk 1.0.16? https://github.com/OpenKinect/libfreenect2/releases/download/v0.2.0/libfreenect2-0.2.0-usbdk-vs2015-x64.zip

wired365, 2017-03-27:
It looks like running usbdk 1.0.16 with the protonect.exe binary you supplied worked just fine. The window popped up with all 4 streams no problem. Does this has something to do with how I compiled the protonect.exe?

xlz, 2017-03-28:
It's a problem between libusb and usbdk and windows driver model.

You have these tests
* newest libusb + usbdk 1.0.17 = LIBUSB_ERROR_OTHER
* newest libusb + usbdk 1.0.16 = bulk transfer failed LIBUSB_ERROR_TIMEOUT
* newest libusb + libusbk = unable to find that there is any kinect
* libfreenect2's libusb build + usbdk 1.0.16 = works

I don't really know. I know how to test it (bisecting libusb's working version) but don't have Windows 10 to test.

To just have something working you can start by editing https://github.com/OpenKinect/libfreenect2/blob/master/depends/install_libusb_vs2015.cmd

Change it to something like
```
git clone https://github.com/libusb/libusb.git libusb_src || exit /b
cd libusb_src
git checkout v1.0.21-rc1
```
`v1.0.21-rc1` just to choose a certain earlier version. Jump between various rc's or use git bisect to find out specific commit that breaks it. Whatever.

wired365, 2017-03-29:
So I apologize in advance for potentially wasting your time. I started getting some strange errors and going backwards so I decided to start at the newest version of everything again and work back through libusb versions like you suggested. When I went to retest old failed combinations I found that libusb 1.0.21 and usbdk 1.0.17 now seem to work on my system. I don't know why it now works because I tested that combination multiple times before with no success. Anyway protonect compiled and ran, I got all of the camera feeds and I can now start development. Thanks for all of your help.

xlz, 2017-03-29:
I saw usbdk 1.0.17 have some weird issues that required rebooting or reinstallation to work. Probably related.

bn-dignitas, 2017-04-11:
I realize this has been closed for a few days, but wanted to add another data point as I saw the same error with USBDK 1.0.17 (haven't yet tested with 16).  The protonect example initially ran fine, but as I tried different CLI options it would fail with LIBUSB_ERROR_OTHER.  This seems to be some sort of timing issue.  If I introduce any sort of delay (i.e. sleep or processing) after the freenect2.enumerateDevices() call in Protonect.cpp, it will be successful.  My best guess is that the device is not completely freed up before open is called on it again later and giving it a bit more time helps it out.  As I said before, a data point and not really a long term solution.

When the failure happens the VS debugger dumps some extra info:
`
First-chance exception at 0x00007FFDB1743C58 in Protonect.exe: Microsoft C++ exception: UsbDkDriverFileException at memory location 0x00000095014FDA30.
Driver file operation error. DeviceIoControl failed (The system cannot find the file specified.  Error code = 2)
`
Cheers.

xlz, 2017-04-12:
> The protonect example initially ran fine, but as I tried different CLI options it would fail

Since your observation implied something related to timing, how long did you wait to run it a second time to have the error? Would it work if you wait longer?

> VS debugger dumps

This is helpful, but do you have the full stack trace while you're at it?

bn-dignitas, 2017-04-12:
To clarify I didn't wait between runs, I actually just added the following code:
`libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(250));`

right before the call to:
`dev = freenect2.openDevice(serial, pipeline);`

This creates a delay between the call to enumerateDevices and openDevice.

As far as a stack, that was all that the console output.  I believe the error occurs on the call to libusb_open at libfreenect2.cpp:1009.

I've already rolled back to usbdk 1.0.16 based on your observations and I haven't seen the issue again yet, so not sure how much it's worth your time looking further into this.

xlz, 2017-04-12:
But you had it

> initially ran fine

So presumably there would be a way to revert it to the initial state and have it run fine again?

If the initial run was fine but subsequent runs were not ok it means insufficient cleanup in the usb stack. If you can pinpoint the steps needed to reset it to the initial state it may reveal where exactly the problem is.

How repeatable is this error? IIUC after the first successful run, you never had it running again until you added this delay, and if you remove it it will always fail with LIBUSB_ERROR_OTHER. Is this what you observed?

bn-dignitas, 2017-04-12:
Sorry for the confusing wording.  I had 100% success when running protonect.exe with no CLI options and 100% failure when specifying gl or cpu as options.  Debugging/tracing the code, the only difference I saw when using the gl option is when the OpenGLPacketPipeline is created.  Using the debugger to step through when passing the gl option also resulted in 100% success, leading me to believe there was a timing issue.  Running the debugger without stopping, or at least breaking after the previously mentioned libusb_open call would result in an error.  I added a short sleep before the open call, which resulted again in 100% success with both default options and CLI options of gl and cpu

Not to confuse things, but I did have one case after rolling back to 1.0.16 where I ran protonect immediately(< 2 seconds) after killing it and it did not find the kinect (was NOT the LIBUSB_ERROR_OTHER code).  I ran a few seconds later and things were fine.  I tried to reproduce that, but have been unable to do so.

Let me know if you need further clarification.

xlz, 2017-04-12:
Can you test something?

Add the following code here https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L463
```diff
            if(r == LIBUSB_SUCCESS)
            {
              unsigned char buffer[1024];
              r = libusb_get_string_descriptor_ascii(dev_handle, dev_desc.iSerialNumber, buffer, sizeof(buffer));
              libusb_close(dev_handle);
              r = libusb_open(dev, &dev_handle);
+             if(r > LIBUSB_SUCCESS) { LOG_ERROR << "libusb_open: " << PrintBusAndDevice(dev, r); }
+             libusb_close(dev_handle);

              if(r > LIBUSB_SUCCESS)

```
Second test:
```diff
            if(r == LIBUSB_SUCCESS)
            {
              unsigned char buffer[1024];
              r = libusb_get_string_descriptor_ascii(dev_handle, dev_desc.iSerialNumber, buffer, sizeof(buffer));
              libusb_close(dev_handle);
+             libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(250));
              r = libusb_open(dev, &dev_handle);
+             if(r > LIBUSB_SUCCESS) { LOG_ERROR << "libusb_open: " << PrintBusAndDevice(dev, r); }
+             libusb_close(dev_handle);

              if(r > LIBUSB_SUCCESS)

```
This is to verify if the problem is something about usbdk not synchronously cleaning up its internal state within libusb_close().

bn-dignitas, 2017-04-12:
The first test results in this error on the open:
`First-chance exception at 0x00007FFEC5F83C58 in Protonect.exe: Microsoft C++ exception: UsbDkDriverFileException at memory location 0x000000900C59D3F0.
Driver file operation error. DeviceIoControl failed (The system cannot find the file specified.  Error code = 2)`
and then this on the close:
`First-chance exception at 0x00007FFEC2232043 (libusb-1.0.dll) in Protonect.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
Unhandled exception at 0x00007FFEC2232043 (libusb-1.0.dll) in Protonect.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.`

Your log is never printed because r is -99

The second test does not crash, but no kinect devices are found (which maybe you expected?):
`failed to get serial number of Kinect v2: @2:2`

xlz, 2017-04-12:
I expected the second test to work just as how it worked with your time delay. Nevertheless, this is enough for an upstream report. https://github.com/daynix/UsbDk/issues/43

bn-dignitas, 2017-04-13:
Additional data point.  I saw something similar on another Windows 10 machine with usbdk 1.0.16, using libfreenect2 in our code.  The difference is that a machine reboot was required to connect to the Kinect after things error out.  Once again, a short sleep after the libusb_close call in enumerateDevices serves as a workaround for the issue.

xlz, 2017-04-21:
Apparently I'm really supposed to wait for a bit after libusb_close() even though libusb API doesn't say so.

I've put out this PR #830. Please see if it works for you.
@wired365 @bn-dignitas

closed 2017-04-24 by xlz

wired365, 2017-09-21:
I want to reopen this issue, as it seems a recent Windows 10 update has caused issues with the USB connection. I'm still getting the usual amount of dropped frames, which is not a problem, but now I am receiving invalid IR and Color camera parameters from the `getIrCameraParams()` and `getColorCameraParams()`. The full issue is detailed in a new issue I posted #912

PR #830 (2017-04-24): usb: Retry a few times with libusb_open()
On some platforms (UsbDk) libusb_open() will return errors for a
short while immediately after libusb_close(). Retry a few times.

Also, fix libusb device reference handling across open and close.

Fix #812.
commit 6a63380e (2017-04-24): usb: Retry a few times with libusb_open()

On some platforms (UsbDk) libusb_open() will return errors for a
short while immediately after libusb_close(). Retry a few times.

Also, fix libusb device reference handling across open and close.

Fix #812.