OpenKinect/libfreenect2 #814: How can I build this on Centos 7.0?

fdp0525, 2017-03-22:
How can I build this on Centos7.0?

smokhov, 2017-03-22:
@fdp0525 we succeeded building it on CentOS 6.x (#711) as a part of the [OpenISS project](https://github.com/OpenISS/OpenISS); it should be easier to do so for CentOS 7.x, but we haven't gotten around to test 7.x yet. We used `kernel-ml`. You can try to reproduce our steps from [`el6.sh`](https://github.com/OpenISS/OpenISS/blob/master/src/scripts/dependencies/el6.sh) for CentOS 7.x for `libfreenect2` part manually and post results. We plan to have `el7.sh` at some point there.

xlz, 2017-03-28:
Individual dependencies are all listed in the Linux section of README. I don't know specific packaging structure of CentOS but you can find the required packages yourself with the list.

closed 2017-03-28 by xlz