OpenKinect/libfreenect2 #817: USB device registration is blocked while running kinect2 over libfreenect2

doge-of-the-day, 2017-03-27:
Has anybody an idea why the registration of other USB devices is blocked, when the kinect2 is accessed via libfreenect2? It may be related to 'libusb', but we are not sure about that at the moment. 
Has anybody else this issue on other platforms or operating systems?

Version, Platform, and Hardware ... :

1.  version : 1f18c24
2.  platform : ubuntu 14.04 LTS - 4.4.0-64-generic
3.  driver : xhci_hcd
4.  chipset: Intel Corporation Q87 Express - USB xHCI
5.  libraries: libusb-1.0.21

Steps to Reproduce:

1. start Protonect (for example)
2. plug in any USB device
3. USB device isn't recognized by the system.

xlz, 2017-03-28:
`dmesg`

AdrianZw, 2017-03-31:
Here is the output of dmesg (doge-of-the-day and I are colleagues and are working together on this project):

```
[  142.717026] usb 4-1: new SuperSpeed USB device number 2 using xhci_hcd
[  142.734691] usb 4-1: New USB device found, idVendor=045e, idProduct=02d9
[  142.734698] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  142.734702] usb 4-1: Product: NuiSensor Adaptor      
[  142.734706] usb 4-1: Manufacturer: Microsoft Corporation  
[  142.736560] hub 4-1:1.0: USB hub found
[  142.737206] hub 4-1:1.0: 1 port detected
[  142.924754] usb 3-1: new high-speed USB device number 8 using xhci_hcd
[  143.055130] usb 3-1: New USB device found, idVendor=045e, idProduct=02d9
[  143.055132] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  143.055134] usb 3-1: Product: NuiSensor Adaptor      
[  143.055135] usb 3-1: Manufacturer: Microsoft Corporation  
[  143.055834] hub 3-1:1.0: USB hub found
[  143.056082] hub 3-1:1.0: 1 port detected
[  144.272705] usb 4-1.1: new SuperSpeed USB device number 3 using xhci_hcd
[  144.289413] usb 4-1.1: New USB device found, idVendor=045e, idProduct=02d8
[  144.289415] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  144.289417] usb 4-1.1: Product: Xbox NUI Sensor
[  144.289418] usb 4-1.1: Manufacturer: Microsoft
[  144.289419] usb 4-1.1: SerialNumber: 500005443142
[  145.960767] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[  146.390092] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 9 ep 2 with no TDs queued?
[  158.783553] usb 3-10: new high-speed USB device number 9 using xhci_hcd
[  158.916936] usb 3-10: New USB device found, idVendor=18a5, idProduct=0302
[  158.916939] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  158.916940] usb 3-10: Product: STORE N GO
[  158.916941] usb 3-10: Manufacturer: Verbatim
[  158.916942] usb 3-10: SerialNumber: CF0059FF1D212C43
[  158.917596] usb 3-10: Not enough bandwidth for new device state.
[  158.917600] usb 3-10: can't set config #1, error -28
```

xlz, 2017-03-31:
> usb 3-10: Not enough bandwidth for new device state.

https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#check-dmesg-after-running-protonect
https://github.com/OpenKinect/libfreenect2/wiki/Linux-USB-Notes#not-enough-bandwidth-for-new-device-state

You need to check your USB hardware  configuration. How is it connected to PCIe bus? What is the bandwidth of the PCIe bus? Any confusion between USB 2.0 and 3.0?

doge-of-the-day, 2017-04-03:
Our system uses an iBase MI981AF mainboard, seems that all external connectors are all on the USB3 same Bus. So if bandwidth is the problem, we will try setup the internal connectors for USB2. Most probably this will solve the problem. 
Thank you! ;)

closed 2017-04-03 by doge-of-the-day