OpenKinect/libfreenect2 #818: Compile error C++11 CUDA

caselitz, 2017-03-27:
Hi,

I have problems with compiling libfreenect2 using
- Ubuntu 14.04
- cmake 2.8.12.2
- gcc 4.8.4.
- CUDA 8.0

when I set ENABLE_CXX11 to ON. I get the following errors:

```
Scanning dependencies of target generate_resources_tool
[  2%] Building CXX object CMakeFiles/generate_resources_tool.dir/tools/generate_resources.cpp.o
Linking CXX executable bin/generate_resources_tool
[  2%] Built target generate_resources_tool
[  5%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
nvcc warning : The 'compute_20', 'sm_20', and 'sm_21' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: identifier "nullptr" is undefined

/usr/lib/gcc/x86_64-linux-gnu/4.8/include/stddef.h(432): error: expected a ";"

/usr/include/x86_64-linux-gnu/c++/4.8/bits/c++config.h(190): error: expected a ";"

/usr/include/c++/4.8/exception(63): error: expected a ";"

/usr/include/c++/4.8/exception(68): error: expected a ";"

/usr/include/c++/4.8/exception(76): error: expected a ";"

/usr/include/c++/4.8/exception(83): error: expected a ";"

/usr/include/c++/4.8/exception(93): error: expected a "{"

/usr/include/c++/4.8/bits/exception_ptr.h(64): error: function "std::current_exception" returns incomplete type "std::__exception_ptr::exception_ptr"

/usr/include/c++/4.8/bits/exception_ptr.h(64): error: expected a "{"

/usr/include/c++/4.8/bits/exception_ptr.h(79): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(81): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(82): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(84): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: declaration is incompatible with previous "std::current_exception"
(64): here

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(86): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(87): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(90): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(92): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(95): error: incomplete type is not allowed

/usr/include/c++/4.8/bits/exception_ptr.h(95): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(116): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(126): error: expected a ";"

/usr/include/c++/4.8/bits/exception_ptr.h(143): error: use of a local type to declare a function

/usr/include/c++/4.8/bits/exception_ptr.h(144): error: expected a ";"
[... 144 lines cut ...]
/usr/include/c++/4.8/cmath(375): error: expected a ";"

/usr/include/c++/4.8/cmath(406): error: inline specifier allowed on function declarations only

/usr/include/c++/4.8/cmath(406): error: variable "std::constexpr" has already been defined

/usr/include/c++/4.8/cmath(406): error: expected a ";"

/usr/include/c++/4.8/cmath(443): error: inline specifier allowed on function declarations only

/usr/include/c++/4.8/cmath(443): error: variable "std::constexpr" has already been defined

/usr/include/c++/4.8/cmath(443): error: expected a ";"

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_000064b8_00000000-7_cuda_kde_depth_packet_processor.cpp1.ii".
Compilation terminated.
CMake Error at cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o.cmake:264 (message):
  Error generating file
  /home/caselitz/software/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

caselitz, 2017-03-27:
Adding the following line to the `IF(ENABLE_CXX11)` case in the main CMakeLists.txt resolved the issue:

`    SET(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11")`

caselitz, 2017-03-27:
related: https://github.com/code-iai/iai_kinect2/pull/381

closed 2017-03-28 by xlz

InstantWindy, 2020-04-30:
@caselitz Hi.I have the same error. You say " in the main CMakeLists.txt , adding a line code",which CMakeLists.txt ?Where is the path of the CMakeLists.txt? Thanks

caselitz, 2020-04-30:
Hi @InstantWindy, I saw that your issue is actually [this one](https://github.com/colmap/colmap/issues/856) and had a quick look. My guess is that `nvcc`, the NVIDIA compiler that compiles CUDA (.cu) files, did not get the compile flags from CMake to enable c++11 support (as mentioned by @tsattler). Try to insert the line `set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11")` in the top level CMakeLists.txt of the colmap project. I'd recommend to add it somewhere [here](https://github.com/colmap/colmap/blob/dev/CMakeLists.txt#L197)...

InstantWindy, 2020-04-30:
I inserted the line set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11") in the top level CMakeLists.txt of the colmap project. but it didn't work.

laoyandujiang, 2020-05-06:
@caselitz  I meet a simlar problem, and I solve it with the solution introduced by @InstantWindy. Specifically, I insert the line "set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++14")" (not c++11) in the line 197. But my error is a bit different from yours. My first error message is: /usr/lib/gcc/x86_64-linux-gnu/5/include/stddef.h(436): error: identifier "nullptr" is undefined. And this is the fifth message in your error.

Ironbrotherstyle, 2020-05-07:
> I inserted the line set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11") in the top level CMakeLists.txt of the colmap project. but it didn't work.

Hi，I added the command `set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11")` in the row of 197

But still got errors at the end of make (about 96%)

[ 96%] Building NVCC (Device) object src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_patch_match_cuda.cu.o
[ 96%] Building NVCC (Device) object src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_gpu_mat_prng.cu.o
[ 96%] Building NVCC (Device) object src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_gpu_mat_ref_image.cu.o

/usr/include/c++/4.9/bits/stl_algo.h(3587): error: expression must have a constant value

/usr/include/c++/4.9/bits/stl_algo.h(3587): error: expression must have a constant value

1 error detected in the compilation of "/tmp/tmpxft_000169e7_00000000-5_gpu_mat_prng.cpp4.ii".
1 error detected in the compilation of "/tmp/tmpxft_000169ef_00000000-5_gpu_mat_ref_image.cpp4.ii".
CMake Error at colmap_cuda_generated_gpu_mat_prng.cu.o.Release.cmake:282 (message):
  Error generating file
  /home/xxx/Projects/xxx/colmap/build/src/CMakeFiles/colmap_cuda.dir/mvs/./colmap_cuda_generated_gpu_mat_prng.cu.o

make[2]: *** [src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_gpu_mat_prng.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMake Error at colmap_cuda_generated_gpu_mat_ref_image.cu.o.Release.cmake:282 (message):
  Error generating file
  /home/xxx/Projects/xxx/colmap/build/src/CMakeFiles/colmap_cuda.dir/mvs/./colmap_cuda_generated_gpu_mat_ref_image.cu.o

make[2]: *** [src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_gpu_mat_ref_image.cu.o] Error 1
/usr/include/c++/4.9/bits/stl_algo.h(3587): error: expression must have a constant value

1 error detected in the compilation of "/tmp/tmpxft_000169df_00000000-5_patch_match_cuda.cpp4.ii".
CMake Error at colmap_cuda_generated_patch_match_cuda.cu.o.Release.cmake:282 (message):
  Error generating file
  /home/xxx/Projects/xxx/colmap/build/src/CMakeFiles/colmap_cuda.dir/mvs/./colmap_cuda_generated_patch_match_cuda.cu.o

make[2]: *** [src/CMakeFiles/colmap_cuda.dir/mvs/colmap_cuda_generated_patch_match_cuda.cu.o] Error 1
make[1]: *** [src/CMakeFiles/colmap_cuda.dir/all] Error 2
make: *** [all] Error 2

Did you have this problem? Please tell me if you have any suggestions. Thank you  so much.

Ironbrotherstyle, 2020-05-07:
> Adding the following line to the `IF(ENABLE_CXX11)` case in the main CMakeLists.txt resolved the issue:
> 
> ` SET(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11")`

Hi，I added the command set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++11") in the row of 197

But still got errors at the end of make (about 96%), as detailed on this page. Your any suggestion is much appreciated. Thank you.

zhangxiaoya, 2020-05-19:
> @caselitz I meet a simlar problem, and I solve it with the solution introduced by @InstantWindy. Specifically, I insert the line "set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++14")" (not c++11) in the line 197. But my error is a little different from yours. My first error message is: /usr/lib/gcc/x86_64-linux-gnu/5/include/stddef.h(436): error: identifier "nullptr" is undefined. And this is the fifth message in your error.

works for me, thanks

joseporiolayats, 2022-03-12:
> @caselitz I meet a simlar problem, and I solve it with the solution introduced by @InstantWindy. Specifically, I insert the line "set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -std=c++14")" (not c++11) in [...]

@laoyandujiang your proposed solution worked for me, was stuck for so many time.. thank you!!

commit e1151dfa (2017-03-28): cmake: Update CUDA C++11 build flags

nvcc can't use only -Xcompiler -std=c++11, it needs -std=c++11 in the whole
pipeline.

The real fix is in CMake 3.3.

https://github.com/Kitware/CMake/commit/99abebdea01b9ef73e091db5594553f7b1694a1b

Fix #818.