OpenKinect/libfreenect2 #820: make command fails when installing on OSX

jshaw, 2017-03-29:
I'm going through the steps to install and when I get to the build section and execute I'm good up until the `make` command.

First, this is the output when running `cmake ..`. Maybe seeing the feature list might help. Maybe this would help.

```
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   FLAG_SEARCH_PATHS_FIRST    1
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    yes
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   freenect2
   freenect2-openni2
```

The error I get during running `make` is the following error:

```
[  4%] Built target generate_resources_tool
[  7%] Building NVCC (Device) object CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o
nvcc fatal   : The version ('80000') of the host compiler ('Apple clang') is not supported
CMake Error at cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o.cmake:207 (message):
  Error generating
  /Users/####/Documents/union/####/kinect/libfreenect2/build/CMakeFiles/cuda_compile.dir/src/./cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/cuda_compile.dir/src/cuda_compile_generated_cuda_kde_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

I am running OSX 10.12.3 and AMD Radeon R9 M370X 2048 MB / Intel Iris Pro 1536 MB.

It seems that there have been a few people who have got it running on intel processor (http://szymonkaliski.com/blog/2017-01-18-kinect-2-osx/) but I'm a bit at a loss as to what the issue might be.

xlz, 2017-03-29:
```
NVIDIA CUDA TOOLKIT 8.0
RN-06722-001 _v8.0 | February 2017
Release Notes for Windows, Linux, and Mac OS
New Features
CUDA Tools
‣ CUDA Compilers. The CUDA compiler now supports Xcode 8.2.1.
```
So, your CUDA version and Xcode version?

xlz, 2017-03-29:
I wonder if CUDA supports `AMD Radeon R9 M370X`???

xlz, 2017-04-21:
I suppose this is a CUDA support issue for Xcode.

closed 2017-04-21 by xlz