OpenKinect/libfreenect2 #821: How to get the raw phase image

wangsen1312, 2017-03-30:
The Kinect V2 should have 3*3 phase image for each depth image. How to get them?

I have tried to use the DumpPacketPipeline to get the "rawest" data and decode it myself , but the IR frame seems not right.

The code I used is:

 libfreenect2::Freenect2 freenect2;
libfreenect2::Freenect2Device *dev = nullptr;
libfreenect2::DumpPacketPipeline *pipeline = nullptr; 

	if (freenect2.enumerateDevices() == 0){
		cout << "no device connected!" << endl;
		return -1;
	}
	string serial = freenect2.getDefaultDeviceSerialNumber();
        int depthProcessor = Processor::cpu;
	if (depthProcessor == Processor::cpu)
	{
		if (!pipeline)
		pipeline = new libfreenect2::DumpPacketPipeline();
	}
	else{
		cout << "failure using other processor (OPENCL or CUDA)!" << endl;
		return -1;
	}

	if (pipeline){
		dev = freenect2.openDevice(serial, pipeline);//! [open]
	}
	else {
		dev = freenect2.openDevice(serial);
	}

bool enable_rgb = false;
	bool enable_depth = true;

	int types = 0;
	if (enable_rgb){
		types |= libfreenect2::Frame::Color;
	}
	if (enable_depth){
		types |= libfreenect2::Frame::Ir | libfreenect2::Frame::Depth;
	}
	libfreenect2::SyncMultiFrameListener listener(types);

	libfreenect2::FrameMap frames;

	dev->setColorFrameListener(&listener);
	dev->setIrAndDepthFrameListener(&listener);

	dev->start();

	Mat depthmat, irmat;

	//! [loop start]
	while (!protonect_shutdown)
	{
		listener.waitForNewFrame(frames);

		libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];
		ir->format = libfreenect2::Frame::Raw;
		libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];
		depth->format = libfreenect2::Frame::Raw;  //hope it works

		cv::Mat(ir->height, ir->width, CV_32FC1, ir->data).copyTo(irmat);
		cv::Mat(depth->height, depth->width, CV_32FC1, depth->data).copyTo(depthmat);

		cv::imshow("ir", irmat / 4500.0f);
		cv::imshow("depth", depthmat / 4500.0f);

		protonect_shutdown = protonect_shutdown || (key > 0 && ((key & 0xFF) == 27)); // shutdown on escape

		//! [loop end]
		listener.release(frames);
	}

Someone can help me? Thank you very much

xlz, 2017-03-30:
You did get `DepthPacket` at `depth->data`.

https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#ab4e88c89b3b7ea1735996cc4def22d58a5bdc2d0b1113f3630fc502864541fc23

> Raw 	Raw bitstream. 'bytes_per_pixel' defines the number of bytes.

wangsen1312, 2017-03-31:
@xlz   
Thank you for your replying.
from DumpPacketPipeline,  the raw IR and raw depth data have any difference?

floe, 2017-03-31:
The IR/depth frames are calculated from the 9 raw phase images. There's no 1:1 relationship.

wangsen1312, 2017-03-31:
@floe    Thank you for your replying.  
I have seen that the depth image are calculated from the 9 raw images, This is using the ToF principal. I don't know how the ordinary IR images come from? average?

xlz, 2017-03-31:
The 'ordinary IR images' are also computed from the 9 raw images. You will need to read relevant algorithms to understand how it is computed. You can start from here https://github.com/OpenKinect/libfreenect2/blob/master/src/cpu_depth_packet_processor.cpp

xlz, 2017-04-02:
Question is answered.

closed 2017-04-02 by xlz

HIMA-MB02, 2019-01-30:
Which version of OpenCV did you use?