OpenKinect/libfreenect2 #825: Device not found! -- why? please...

limengze126, 2017-04-13:
I tried to connect Kinect with Mac OS. I followed the instructions step by step and no error occurred. However, when I tried to run "bin/Protonect" it always said no device connected.

By the way, when I try to verify XBOX NUI Sensor is under USB 3.0, I surprisingly found out that there is no  Xbox Nui Sensor under NuiSensor Adaptor. (in system information).

Could you please help me with it?

Thanks a lot!
Best,
Mengze Li

xlz, 2017-04-14:
Did you plug Kinect to the adaptor?

limengze126, 2017-04-14:
Thank you for your reply! I plugged Kinect to the adaptor. Later, I tried to unplug the power plug and re-plug, and it works.

Best,
Mengze

closed 2017-04-14 by xlz