OpenKinect/libfreenect2 #826: How to use the color and depth params to achieve color_to_depth() transformation

Spade-J, 2017-04-14:
well there is a method in registration.cpp called depth_to_color(float mx, float my, float& rx, float& ry),and I want to know how to use the params we get to realize the method  color_to_depth(). thanks

Spade-J, 2017-04-14:
I have used the depth_to_color() method to replace the interface MapDepthFrameToColorSpace() in kinect, and now, I want to replace the interface MapColorFrameToDepthSpace() in kinect by the method color_to_depth(), but I haven't found this method, so should I code this by myself? But I don't know how to use this params , they are not matrix

floe, 2017-04-14:
`MapDepthFrameToColorSpace` is from the Microsoft Kinect SDK, so I think you're confusing something here. libfreenect2 is a standalone library that has nothing to do with the "official" MS SDK.

xlz, 2017-04-14:
depth_to_color() is analytic. You can solve the inverse yourself. But it would be inaccurate because it is an approximation in the first place, which is why no implementation is provided.

closed 2017-04-14 by xlz

kidapu, 2017-09-11:
Is it possible to inverse calculation on [this cubic equation](https://github.com/OpenKinect/libfreenect2/blob/master/src/registration.cpp#L92-L102)?

xlz, 2017-09-16:
@kidapu It is possible. And it is not difficult to derive. But as I said above, it would be inaccurate.

kidapu, 2017-09-24:
@xlz

 Can I tell some hints (keyword or url) to derive ?  I want to try.

I want to convert RGB 2D coordinate -> World 3D coordinate .

xlz, 2017-09-24:
Here's the implementation for a similar problem. Just the generaly impression don't look too closely. https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L156

You can use Newton's method (many textbooks for this) to numerically compute the solution. Analytical methods are probably more work than numeric methods. Newton's method requires the Jacobian matrix, which you can derive with Wolfram Alpha.