OpenKinect/libfreenect2 #827: wrong cmake for windows with Strawberry Perl pkg-config?

hamster1979, 2017-04-14:
```
- Detecting CXX compile features                                                                                        
- Detecting CXX compile features - done                                                                                 
- using tinythread as threading library                                                                                 
- Found PkgConfig: D:/Strawberry/perl/bin/pkg-config                                                                    
epends DIR: D:/work/projects/3dsensor/distr/libfreenect2-master/libfreenect2-master/depends                             
et environment :D:/work/projects/3dsensor/distr/libfreenect2-master/libfreenect2-master/depends/libusb/lib/pkgconfig    
- Checking for module 'libusb-1.0'                                                                                      
-                                                                                                                       
Make Error at D:/Program Files/CMake/share/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):                         
 A required package was not found                                                                                       
all Stack (most recent call first):                                                                                     
 D:/Program Files/CMake/share/cmake-3.8/Modules/FindPkgConfig.cmake:589 (_pkg_check_modules_internal)                   
 cmake_modules/FindLibUSB.cmake:25 (PKG_CHECK_MODULES)                                                                  
 CMakeLists.txt:93 (FIND_PACKAGE)                                                                                       
```                                                            
Because i have Strawberry Perl installed in windows with pkg-config, in FindLibUSB.cmake 
```
IF(PKG_CONFIG_FOUND)
  IF(DEPENDS_DIR) #Otherwise use System pkg-config path
    SET(ENV{PKG_CONFIG_PATH} "$ENV{PKG_CONFIG_PATH}:${DEPENDS_DIR}/libusb/lib/pkgconfig")
  ENDIF()
``` 
sets wrong paths /libusb/lib/pkgconfig

xlz, 2017-04-14:
`./libusb/lib/pkgconfig` is an arbitrary path in which you're supposed to put the pkgconfig file of libusb.

closed 2017-04-14 by xlz

hamster1979, 2017-04-14:
IF(PKG_CONFIG_FOUND) is not correct condition for windows build IMHO. Without this all ok (SET (PKG_CONFIG_FOUND 0)).

xlz, 2017-04-14:
If you want to use a POSIX environment (which is what you're using, a MingW distribution), you will have to provide pkgconfig files for libusb. `install_libusb_vs2013.cmd` is not for building libusb under a POSIX environment. You would use `./configure && make && make install` to create libusb installation with pkgconfig files, which is the standard way to build POSIX packages.