OpenKinect/libfreenect2 #828: Cannot view bigdepth

sairamkolla, 2017-04-15:
Hi,

I'm trying to find depth of a pixel in the rgb image. I tried using bigdepth but I'm not able to view the bigdepth image( all values are inf) 

The code i used is 

`libfreenect2::Frame* bigDepth = new libfreenect2::Frame(1920,1082,4);`
` registration->apply(rgb, depth, &undistorted, &registered,true,bigDepth);`
` cv::Mat bigDepthMat;`
` cv::Mat(bigDepth->height, bigDepth->width, CV_32FC1, bigDepth->data).copyTo(bigDepthMat);`
` cv::imshow("registeredInv", bigDepthMat/4500.0f);`

philipNoonan, 2017-04-15:
This worked for me, ugly code, apologies. Honestly, please don't judge.

registration->apply(rgb, depth, &undistorted, &registered, true, &bigDepth, colorDepthIndex);
cv::Mat temp1 = cv::Mat(1082, 1920, CV_32F, bigDepth.data);
cv::Mat temp2;
temp1.convertTo(temp2, CV_16UC1);
cv::imshow("depth", 50 * temp2);

sairamkolla, 2017-04-15:
Hi @philipNoonan 

I tried your code, but it's showing a complete white image.

philipNoonan, 2017-04-23:
I think you didn't replace the tempMat variable for the bigDepthMat variable. Both your code and my code work fine over here (with the replace fix mentioned).

closed 2017-05-14 by floe

sairamkolla, 2017-05-19:
Hello @philipNoonan ,

I've made the necessary changes in my code, but still it is giving a black image as output. This is my [code](https://pastebin.com/CxN4KzfW) and [cmakefile](https://pastebin.com/TNfdNw1a). The problem is only in the first iteration. From the second iteration, I'm able to view all outputs. Please let me know  the mistake I've been committing so far. I'm assuming proper initialisation is not done for bigdepth.

Regards,
sairam