OpenKinect/libfreenect2 #829: `Could NOT find TegraJPEG` and `Linux4Tegra version (21.1.0) is not recognized`

alexge233, 2017-04-18:
Hi,

I'm trying to install on a new TX2 which has JetPack 3.0/R27.1 installed.
The complete output of cmake is:

```bash
ubuntu@tegra-ubuntu:~/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.20
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Found JPEG: /usr/lib/aarch64-linux-gnu/libjpeg.so  
-- Found Linux4Tegra 27.1.0
CMake Warning at cmake_modules/FindTegraJPEG.cmake:51 (MESSAGE):
  Linux4Tegra version (27.1.0) is not recognized.
Call Stack (most recent call first):
  CMakeLists.txt:217 (FIND_PACKAGE)

-- Could NOT find TegraJPEG (missing:  TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/aarch64-linux-gnu/libturbojpeg.so.0  
-- Checking for module 'glfw3'
--   No package 'glfw3' found
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing:  OpenCL_LIBRARY OpenCL_INCLUDE_DIR) 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-8.0 (found version "8.0") 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/aarch64-linux-gnu/libusb-1.0.so
 pthread
 /usr/lib/aarch64-linux-gnu/libturbojpeg.so.0
 /usr/local/cuda-8.0/lib64/libcudart_static.a
 -lpthread
 dl
 /usr/lib/aarch64-linux-gnu/librt.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'glfw3'
--   No package 'glfw3' found
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib  
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    no
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/libfreenect2/build
```

I took a peek in `cmake_modules/FiindTegraJPEG.cmake` and saw that lines 36 - 50 check for version up to R24.2. Is it too much trouble to ask for a quick patch to fix with TX2?

Many thanks!

xlz, 2017-04-20:
I don't have a TX2, but I made this patch. Tell us if it works.
```diff
diff --git a/cmake_modules/FindTegraJPEG.cmake b/cmake_modules/FindTegraJPEG.cmake
index 94ec0e7..daf59c4 100644
--- a/cmake_modules/FindTegraJPEG.cmake
+++ b/cmake_modules/FindTegraJPEG.cmake
@@ -47,6 +47,8 @@ IF(TegraJPEG_IS_L4T)
     SET(L4T_GSTJPEG_URL_PART r24_Release_v1.0/24.1_64bit/source)
   ELSEIF(L4T_VER MATCHES ^24.2)
     SET(L4T_GSTJPEG_URL_PART r24_Release_v2.0/BSP)
+  ELSEIF(L4T_VER MATCHES ^27.1)
+    SET(L4T_GSTJPEG_URL_PART r27_Release_v1.0/BSP)
   ELSE()
     MESSAGE(WARNING "Linux4Tegra version (${L4T_VER}) is not recognized.")
     SET(TegraJPEG_L4T_OK FALSE)
@@ -57,6 +59,8 @@ ENDIF()
 IF(TegraJPEG_L4T_OK)
   IF(L4T_VER VERSION_LESS 24.2.0)
     SET(L4T_SRC_FILE "gstjpeg_src.tbz2")
+  ELSEIF(L4T_VER VERSION_EQUAL 27.1.0)
+    SET(L4T_SRC_FILE "r27.1.0_sources.tbz2")
   ELSE()
     SET(L4T_SRC_FILE "sources.tbz2")
   ENDIF()
```

alexge233, 2017-04-21:
I'm afraid this didn't work, output below:

```bash
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.20
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Found JPEG: /usr/lib/aarch64-linux-gnu/libjpeg.so  
-- Found Linux4Tegra 27.1.0
-- Downloading r27.1.0_sources.tbz2...
-- Extracting r27.1.0_sources.tbz2...
CMake Warning at cmake_modules/FindTegraJPEG.cmake:95 (MESSAGE):
  Failed to unpack gstjpeg_src.tbz2: CMake Error: Problem with
  archive_read_open_file(): Failed to open
  '/home/ubuntu/libfreenect2/depends/source/gstjpeg_src.tbz2'

  CMake Error: Problem extracting tar:
  /home/ubuntu/libfreenect2/depends/source/gstjpeg_src.tbz2

Call Stack (most recent call first):
  CMakeLists.txt:217 (FIND_PACKAGE)

-- Could NOT find TegraJPEG (missing:  TegraJPEG_INCLUDE_DIRS TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/aarch64-linux-gnu/libturbojpeg.so.0  
-- Checking for module 'glfw3'
--   No package 'glfw3' found
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing:  OpenCL_LIBRARY OpenCL_INCLUDE_DIR) 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-8.0 (found version "8.0") 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/aarch64-linux-gnu/libusb-1.0.so
 pthread
 /usr/lib/aarch64-linux-gnu/libturbojpeg.so.0
 /usr/local/cuda-8.0/lib64/libcudart_static.a
 -lpthread
 dl
 /usr/lib/aarch64-linux-gnu/librt.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'glfw3'
--   No package 'glfw3' found
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib  
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    no
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/libfreenect2/build
```

xlz, 2017-04-21:
Can you try again with this commit? https://github.com/xlz/libfreenect2/commit/39fae28e6b3c01b1f9e2d07fc914dbf042d02154

alexge233, 2017-04-24:
Hi @xlz yes this solved the issue! Many thanks! Runs much better and faster on TX2 than on a i7-6500U!

closed 2017-04-24 by alexge233