OpenKinect/libfreenect2 #831: bulk transfer failed: LIBUSB_ERROR_TIMEOUT with multiple kinects

dimitar-rakov, 2017-04-21:
Hi, 

I'm trying to work simultaneously with 2 kinects  in Ubuntu 14.04 with Kernel 4.4, but I always receive **bulk transfer failed: LIBUSB_ERROR_TIMEOUT**  when I start a second  sensor. One by one the sensors work  fine. The first one is connected through an on-board Intel USB3 controller  and second one through an external Renesas controller (uPD720201) on PCIe x1. 

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
`commit 81c894ca32b00781058b45b959a4376164e4d39a`

2. `uname -a`
`4.4.0-72-generic #93~14.04.1-Ubuntu SMP Fri Mar 31 15:05:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

3. `lsusb -t`
`/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 5, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 5, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 7: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 9: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 10: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M`

4. `lspci -nn`
`00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
00:1c.1 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 [8086:8c12] (rev d5)
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
00:1c.4 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 [8086:8c18] (rev d5)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation C220 Series Chipset Family H81 Express LPC Controller [8086:8c5c] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 750] [10de:1381] (rev a2)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:0fbc] (rev a1)
03:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller [1912:0014] (rev 03)
04:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)
05:00.0 USB controller [0c03]: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller [1b21:1142]
`

Steps to Reproduce:

1. ./bin/Protonect cuda 011296264147  (same without cuda prefix)
2. ./bin/Protonect cuda 003722754847  (same without cuda prefix)

Actual Results:

`Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 750 @ 1137MHz Memory 1991MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 19 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:3 with serial 003722754847
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 011296264147
[Info] [Freenect2Impl] found 2 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
`

Additional Information:

I tried already to increase the USBFS buffer size up to 256 and I also I made tests with different Kernels (3.19.0-80 and 4.2.0-42), both without any success. I have attached a log file with result produced by  LIBUSB_DEBUG=4 ./bin/Protonect.
[log.txt](https://github.com/OpenKinect/libfreenect2/files/947454/log.txt)

Could the problem be, that  one of usb3 controllers is build in? Has anybody had a similar experience?

Thanks a lot!

xlz, 2017-04-21:
Here is how to read the log.txt. The first error in it is this:
```
[ 0.330203] [00000e47] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 1835008
[ 0.330206] [00000e47] libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
[ 0.330208] [00000e47] libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
[ 0.330209] [00000e47] libusb: debug [disarm_timerfd]
[ 0.330210] [00000e47] libusb: debug [libusb_free_transfer] transfer 0x249a658
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
```
Errno 12 is ENOMEM, so this is classic out of memory issue. See https://github.com/OpenKinect/libfreenect2/issues/458#issuecomment-155102173.

If you've really set `usbfs_memory_mb` to 256 and it still didn't work, we will need more information from `dmesg` to tell the problem.

dimitar-rakov, 2017-04-23:
The log file, which I uploaded, is made with `usbfs_memory_mb` equal to 64, but as I said I had the same issue by 256. Tomorrow I will make  again a test for both cases and I'm going to provide you dmesg information.

dimitar-rakov, 2017-04-24:
I managed to fix it. The size of `usbfs_memory_mb` was not set correctly as you suggested. I used the tip given by #807 to set it to 256. 

Thank you for your help!

closed 2017-04-24 by xlz