OpenKinect/libfreenect2 #833: Why install `libgl1-mesa-dri-lts-vivid` on Ubuntu 14.04 Trusty Tahr?

ksze, 2017-04-25:
Could you explain a little bit *why* we should install `libgl1-mesa-dri-lts-vivid`?

I'm trying to use libfreenect2 on both Intel x64 and the NVIDIA Jetson TK1 platforms, both running Ubuntu 14.04 Trusty (actually L4T 21.5 on the Jetson TK1).

So is there any particular reason I should install *specifically* `libgl1-mesa-dri-lts-vivid`?

A newer Hardware Enablement (HWE) stack is available nowadays, based on the same kernel as Ubuntu Xenial, which comes with the newer `libgl1-mesa-dri-lts-xenial` package.

I'm wondering if anybody has already tried the latest HWE stack, and whether that breaks things in known, but non-obvious ways.

floe, 2017-04-25:
Rule of thumb: the newer the kernel, the better. So no, there's no specific reason I know for using exactly the vivid dev package - give the newer ones a try and let us know if it works for you. For pure OpenCL, I'm not sure if you need the mesa packages anyway.

xlz, 2017-04-25:
There is a bug in libgl1-mesa-dri. -vivid or newer resolves it. It's easy to break package dependency when upgrading this single package so -vivid seemed a safe choice.

The viewer isn't affected. If you don't use OpenGL depth processing you don't need it.

closed 2017-04-25 by xlz