OpenKinect/libfreenect2 #834: When/why would *PacketProcessor.ready() ever return false?

ksze, 2017-04-25:
I'm attempting to improve the performance and stability of libfreenect2 on Jetson TK1. The biggest issue is burst of "skipping depth packet".

So I dug through the code and found out that "skipping depth packet" happens when `process_->ready()` returns `false` at https://github.com/OpenKinect/libfreenect2/blob/6a63380ece335e07584ed55fcf7a0dd8b7fedf41/src/depth_packet_stream_parser.cpp#L112. However, I'm unable to find any override of the `ready()` method that would return `false`. I grep'ed the codebase and all I see is this (https://github.com/OpenKinect/libfreenect2/blob/6a63380ece335e07584ed55fcf7a0dd8b7fedf41/include/internal/libfreenect2/packet_processor.h#L51):

    virtual bool ready() { return true; }

What am I missing?

xlz, 2017-04-25:
https://github.com/OpenKinect/libfreenect2/blob/master/include/internal/libfreenect2/async_packet_processor.h#L68

xlz, 2017-04-26:
What do you want to achieve? I have fairly extended experience with performance on TK1.

ksze, 2017-04-26:
I'm trying to achieve stable 30 FPS throughput.

I have a Jetson TK1 board flashed with L4T 21.5, so the CPUs are limited to ~2.0 GHz, not the full ~2.3 GHz.

If I run `Protonect cuda -noviewer -norgb`, I get bursts of about 5 to 15 lines of "skipping depth packet" from time to time. I'm trying to eliminate these bursts.

I have already disabled a bunch of services that I don't really need:
- anacron
- avahi-cups-reload
- avahi-daemon
- bluetooth
- cron
- cryptdisks
- cryptdisks-udev
- cups-browsed
- cups
- lightdm (which means I don't get a graphical desktop)

Right now, it appears that the dynamic scaling of the GPU frequency is mostly to blame. If I lock the GPU frequency to 612 MHz using `/sys/kernel/debug/clock/override.gbus/rate`, the throughput of Protonect is very stable, but that's still with only depth, no rgb. At 612 MHz, there is still some room until the maximum 852 MHz. I'll check what happens when I enable RGB as well.

However, besides receiving and decoding Kinect v2 packets, we will also try to perform realtime processing of the video data on the Jetson TK1, using OpenCV 3.0 (which means we don't get the benefit of NVIDIA's optimized OpenCV 2.4). I hope we still have enough CPU time and memory bandwidth to spare.

Do you have other suggestions?

xlz, 2017-04-26:
Well you would want to max out both GPU and CPU frequencies. There is no reason not to, except for small batteries.

TK1's CPU is extremely slow. Just a few memcpys will grind it to a halt, so OpenCV 3.0 probably doesn't work. Color registration is on CPU so it's slow too.

ksze, 2017-04-26:
I just tried setting all CPUs and the GPU to their max frequencies.

I can now run `Protonect cuda -noviewer`, without depth-rgb registration (I commented out the relevent lines of code in `Protonect.cpp`. It's quite stable, but I sometimes see error messages like these:
```
mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
mc-err: [mcerr]   status = 0x60000022; addr = 0x821aa000
mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
mc-err: [mcerr]   status = 0x60000022; addr = 0x821aa000
mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
mc-err: [mcerr]   status = 0x60000022; addr = 0x821aa000
mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
mc-err: [mcerr]   status = 0x60000022; addr = 0x830aa000
mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
mc-err: [mcerr]   status = 0x60000022; addr = 0x830aa000
mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
mc-err: Too many MC errors; throttling prints
```
These would coincide with spurious rgb and depth packet drops in Protonect:
```
[Debug] [DepthPacketStreamParser] not all subsequences received 63
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] not all subsequences received 63
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] not all subsequences received 63
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] not all subsequences received 3
```

Do you have any idea what are those MC errors? Are those memory controller errors? They *seem* to disappear after running Protonect for a while, I hope it's just a matter of the TK1 ramping up the memory controller frequency or something?

xlz, 2017-04-26:
I've seen them but I don't know much. They weren't a huge issue. RGB drops are a voluntary result of sync with depth, no rgb data is lost.

ksze, 2017-04-27:
> RGB drops are a voluntary result of sync with depth

@xlz Do you mean you have already implemented the temporal synchronization mentioned here (https://github.com/OpenKinect/libfreenect2/issues/721#issuecomment-291603005)?

Could you point me to the commit that does that?

xlz, 2017-04-27:
No

closed 2017-12-01 by xlz