OpenKinect/libfreenect2 #835: Mac with libusb 1.0.21 cmake error.

pexist, 2017-04-28:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
6a63380 usb: Retry a few times with libusb_open()
2. `uname -a`
Darwin uav-pro.local 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1.  cmake ..

Actual Results:

uav-pro:build pex$ cmake ..
-- using tinythread as threading library
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
CMake Error at /opt/local/share/cmake-3.7/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.7/Modules/FindPkgConfig.cmake:588 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)

Expected Results:

Reproducibility:

Additional Information:

pex$ brew install libusb
Warning: libusb-1.0.21 already installed

streamgao, 2017-04-28:
same issus

xlz, 2017-04-28:
The output of:
`pkg-config --debug 'libusb-1.0 >= 1.0.20'; echo $?`
?

streamgao, 2017-04-28:
```
no output option set, defaulting to --exists
Error printing disabled by default due to use of output options --exists, --atleast/exact/max-version, --list-all or no output option at all. Value of --print-errors: 0
Error printing disabled
Adding virtual 'pkg-config' package to list of known packages
Looking for package 'libusb-1.0'
Looking for package 'libusb-1.0-uninstalled'
Reading 'libusb-1.0' from file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr/local/Cellar/libusb/1.0.21
 Variable declaration, 'prefix' has value '/usr/local/Cellar/libusb/1.0.21'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr/local/Cellar/libusb/1.0.21'
  line>libdir=${exec_prefix}/lib
 Variable declaration, 'libdir' has value '/usr/local/Cellar/libusb/1.0.21/lib'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/local/Cellar/libusb/1.0.21/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
  line>Version: 1.0.21
  line>Libs: -L${libdir} -lusb-1.0
  line>Libs.private: -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
Path position of 'libusb-1.0' is 1
Adding 'libusb-1.0' to list of known packages
0
C02SH6LSG8WP-lm:build streamg$ 
C02SH6LSG8WP-lm:build streamg$ clear

C02SH6LSG8WP-lm:build streamg$ pkg-config --debug 'libusb-1.0 >= 1.0.20'; echo $?
no output option set, defaulting to --exists
Error printing disabled by default due to use of output options --exists, --atleast/exact/max-version, --list-all or no output option at all. Value of --print-errors: 0
Error printing disabled
Adding virtual 'pkg-config' package to list of known packages
Looking for package 'libusb-1.0'
Looking for package 'libusb-1.0-uninstalled'
Reading 'libusb-1.0' from file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr/local/Cellar/libusb/1.0.21
 Variable declaration, 'prefix' has value '/usr/local/Cellar/libusb/1.0.21'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr/local/Cellar/libusb/1.0.21'
  line>libdir=${exec_prefix}/lib
 Variable declaration, 'libdir' has value '/usr/local/Cellar/libusb/1.0.21/lib'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/local/Cellar/libusb/1.0.21/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
  line>Version: 1.0.21
  line>Libs: -L${libdir} -lusb-1.0
  line>Libs.private: -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
Path position of 'libusb-1.0' is 1
Adding 'libusb-1.0' to list of known packages
0
```

xlz, 2017-04-28:
First I want to make sure you haven't actually run `find /usr/local -name '*libusb*' -delete` because that doesn't apply to macos.

This time please attach the log file of `cmake --trace-expand .. 2>log.txt` (clean the build directory first).

streamgao, 2017-04-28:
i did run `find /usr/local -name '*libusb*' -delete because that doesn't apply to macos.`
then I try to reapply the process and got problem saying that the libusb was dropped universally. 
so i deleted the tap/homebrew-core folder, and reinstall libusb.    would that work?

and the result of `cmake --trace-expand .. 2>log.txt` is

```
Running with expanded trace output on.
-- using tinythread as threading library
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Checking for module 'glfw3'
--   Found glfw3, version 3.2.1
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- RPATH set to /usr/local/Cellar/libusb/1.0.21/lib
-- Linking with these libraries: 
 /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.dylib
 pthread
 /System/Library/Frameworks/VideoToolbox.framework
 /System/Library/Frameworks/CoreFoundation.framework
 /System/Library/Frameworks/CoreMedia.framework
 /System/Library/Frameworks/CoreVideo.framework
 /usr/local/opt/jpeg-turbo/lib/libturbojpeg.dylib
 /usr/local/Cellar/glfw/3.2.1/lib/libglfw.dylib
 /System/Library/Frameworks/OpenGL.framework
 /System/Library/Frameworks/OpenCL.framework
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'glfw3'
--   Found glfw3, version 3.2.1
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   FLAG_SEARCH_PATHS_FIRST    1
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    yes
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/streamg/SDEV/libfreenect2/build
```

xlz, 2017-04-28:
No problem then?

streamgao, 2017-04-28:
Yes!!

It works now.  But when I try to run `./bin/Protonect` with kinect2 for xbox plugged in my laptop. Mac osSierra 10.124. 
It couldn't detect the device:
```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!
```

xlz, 2017-04-28:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#verify-kinect-is-indeed-connected-to-a-usb-3-port

streamgao, 2017-04-28:
worked finally!!! thanks!!

closed 2017-04-28 by xlz