OpenKinect/libfreenect2 #836: Expose depth confidence map (KDE)

caselitz, 2017-04-28:
Hi,

thank you @felja633 for providing the implementation of the KDE depth packet processor. I am interested in obtaining pixel-wise confidence values for the depth estimates (Eq. 25 in the paper).

Probably this is mainly a framework / API issue. I wonder if it would make sense to extend the API in a way to allow depth packet processors to expose this information, so that it can be obtained like a frame? (Possible enhancement for 0.3? https://github.com/OpenKinect/libfreenect2/issues/667)

For now, can someone (maybe @xlz) suggest how I can achieve that with a (hacky) workaround? Would that be complicated? I have to admit that I am not too familiar with the internals of libfreenect2.

Right now it is also not possible to set the DepthPacketProcessor::Config via the API, right? Could I do that by including libefreenect2 in my project and use the internal headers?

Thank you,
Tim

xlz, 2017-04-28:
> how I can achieve that with a (hacky) workaround?

Pack two values into one float, well, just use multiplication.

> set the DepthPacketProcessor::Config via the API

https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Freenect2Device.html#ae45b1eead7e5b2e1c8e2259ae55663ff

felja633, 2017-05-02:
The final confidence in eq. 25 is stored in the variable max_val in cuda_kde_depth_packet_processor.cu (or opencl_kde_depth_packet_processor.cpp). If you just want to see the confidence you could easily just set 

depth[ i ] = max_val;

caselitz, 2017-05-02:
> Pack two values into one float, well, just use multiplication.
> 
> Specifically, the paper says 18.75 is the maximum. By millimeter precision you can represent it with integer in 0-18750. So you can encode this with 15 bits, and confidence with the rest of the bits in a 32-bit float.

Thank you, that's a nice hacky workaround. :) I somehow thought I would need to extend the interface...

> https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Freenect2Device.html#ae45b1eead7e5b2e1c8e2259ae55663ff

Sorry, I mixed it up. I meant DepthPacketProcessor::Parameters - I wanted to change the kde_threshold over the API.

> The final confidence in eq. 25 is stored in the variable max_val in cuda_kde_depth_packet_processor.cu (or opencl_kde_depth_packet_processor.cpp). If you just want to see the confidence you could easily just set
> 
> depth[i ] = max_val;

Actually I would like to have both, depth and confidence, so I will try what xlz suggested. But thanks for pointing how to access the confidence!

caselitz, 2017-06-02:
I finally found the time to do that. If someone wants to do the same thing:

On the libfreenect2 side:
`uint16_t d_int = (uint16_t)(d * 65535 / 18750 + 0.5);`
`uint16_t max_val_int = (uint16_t)(max_val * 65535 + 0.5);`
`uint32_t d_max_val_int = d_int << 16 | max_val_int;`
`depth[i] = *reinterpret_cast<float *>(&d_max_val_int);`

On your code side:
`float depth_conf_float = ...`
`uint32_t depth_conf_int = *reinterpret_cast<uint32_t *>(&depth_conf_float);`
`uint16_t depth_int = depth_conf_int >> 16;`
`uint16_t conf_int = depth_conf_int;`
`float depth_float = (float)depth_int * 18750 / 65535;`
`float conf_float = (float)conf_int / 65535;`

closed 2017-06-02 by caselitz