OpenKinect/libfreenect2 #838: segmentation fault when run ./Protonect

Allen-jl, 2017-05-04:
fjl@fjl-Inspiron-7460:~/libfreenect2/build/bin$ ./Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 505434742542
[Info] [Freenect2Impl] found 1 devices
Segmentation fault (core dumped)
--------------------------------------------------------------------------------------------------------------
So how can I fix this problem? thx

SirDifferential, 2017-05-04:
Please run Protonect in gdb (linux) or clang (Mac), by calling `gdb ./Protonect`, type "r",  then include the backtrace by typing "bt" after the segfault. Also include more details about your system (which distro, GPU, etc.)

taseenb, 2017-05-06:
I have the same problem, unless i use the `cl` option. I did `lldb ./Protonect` and this is what i get (i removed hundreds of duplicated lines):

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
objc[62997]: Class GLFWWindowDelegate is implemented in both /usr/local/opt/glfw/lib/libglfw.3.dylib (0x1001d8b08) and /Users/Shared/htdocs/LIBS/libfreenect2/depends/glfw/lib/libglfw.3.dylib (0x1001473f0). One of the two will be used. Which one is undefined.
objc[62997]: Class GLFWApplicationDelegate is implemented in both /usr/local/opt/glfw/lib/libglfw.3.dylib (0x1001d8b80) and /Users/Shared/htdocs/LIBS/libfreenect2/depends/glfw/lib/libglfw.3.dylib (0x100147468). One of the two will be used. Which one is undefined.
objc[62997]: Class GLFWContentView is implemented in both /usr/local/opt/glfw/lib/libglfw.3.dylib (0x1001d8ba8) and /Users/Shared/htdocs/LIBS/libfreenect2/depends/glfw/lib/libglfw.3.dylib (0x100147490). One of the two will be used. Which one is undefined.
objc[62997]: Class GLFWWindow is implemented in both /usr/local/opt/glfw/lib/libglfw.3.dylib (0x1001d8c20) and /Users/Shared/htdocs/LIBS/libfreenect2/depends/glfw/lib/libglfw.3.dylib (0x100147508). One of the two will be used. Which one is undefined.
objc[62997]: Class GLFWApplication is implemented in both /usr/local/opt/glfw/lib/libglfw.3.dylib (0x1001d8c48) and /Users/Shared/htdocs/LIBS/libfreenect2/depends/glfw/lib/libglfw.3.dylib (0x100147530). One of the two will be used. Which one is undefined.
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:5 with serial 164614640847
[Info] [Freenect2Impl] found 1 devices
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
2017-05-06 22:33:19.109427+0100 Protonect[62997:3910438] [General] ERROR: Setting <GLFWContentView: 0x10285a710> as the first responder for window <GLFWWindow: 0x100714f60>, but it is in a different window ((null))! This would eventually crash when the view is freed. The first responder will be set to nil.
(
        0   AppKit                              0x00007fff968a2a3f -[NSWindow _validateFirstResponder:] + 557
        1   AppKit                              0x00007fff9601db1c -[NSWindow _setFirstResponder:] + 31
        2   AppKit                              0x00007fff960c5d9e -[NSWindow _realMakeFirstResponder:] + 406
        3   AppKit                              0x00007fff960c5bb4 -[NSWindow makeFirstResponder:] + 123
        4   libglfw.3.dylib                     0x00000001001d1895 _glfwPlatformCreateWindow + 631
        5   libglfw.3.dylib                     0x00000001001cd430 glfwCreateWindow + 487
        6   libfreenect2.0.2.dylib              0x00000001000b9b6c _ZN12libfreenect226OpenGLDepthPacketProcessorC2EPvb + 188
        7   libfreenect2.0.2.dylib              0x00000001000a9763 _ZN12libfreenect220OpenGLPacketPipelineC2EPvb + 163
        8   libfreenect2.0.2.dylib              0x00000001000b45dd _ZN12libfreenect227createDefaultPacketPipelineEv + 253
        9   libfreenect2.0.2.dylib              0x00000001000b5762 _ZN12libfreenect29Freenect210openDeviceERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE + 18
        10  Protonect                           0x0000000100003679 main + 3961
        11  libdyld.dylib                       0x00007fffadc71235 start + 1
        12  ???                                 0x0000000000000001 0x0 + 1
)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined

Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'run_one_line' is not defined
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[... 7 lines cut ...]
[Info] [Freenect2DeviceImpl] started
device serial: 164614640847
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[... cut ...]
[Info] [DepthPacketStreamParser] 384 packets were lost
Process 62997 stopped
* thread #10, name = 'OpenGL', stop reason = EXC_BAD_ACCESS (code=1, address=0x40)
    frame #0: 0x00007fff93f90f61 GeForceGLDriver`___lldb_unnamed_symbol17132$$GeForceGLDriver + 14
GeForceGLDriver`___lldb_unnamed_symbol17132$$GeForceGLDriver:
->  0x7fff93f90f61 <+14>: movq   0x40(%rax), %rax
    0x7fff93f90f65 <+18>: movslq %esi, %rcx
    0x7fff93f90f68 <+21>: movl   (%rax,%rcx), %eax
    0x7fff93f90f6b <+24>: popq   %rbp
```

xlz, 2017-05-06:
@taseenb https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#abort-trap-6-on-macs-with-nvidia-gpus-using-opengl

taseenb, 2017-05-06:
Thanks @xlz and sorry i had not seen that.

xlz, 2017-05-11:
Merge into #842.

closed 2017-05-11 by xlz