OpenKinect/libfreenect2 #841: Protonect (libfreenect2) is not responding and then closing

blueshark000, 2017-05-07:
I am trying to use Kinect v2 in Ubuntu 14.04 64 bit platform. I am using libfreenect2. I followed the installation procedure and didn't get any error. However when I run ./bin/Protonect, the pop-up window appear but it doesn't respond. After few seconds, this window closed automatically. Below is the screenshot-
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 007676264947
[Info] [Freenect2Impl] found 1 devices
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9839
[Debug] [Freenect2DeviceImpl] status 0x090000: 9839
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 007676264947
device firmware: 4.0.3916.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.4248ms -> ~40.942Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 25.4273ms -> ~39.3279Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
timeout!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
pure virtual method called
terminate called without an active exception

what's the problem, I have already being dealing with it for a few days. 3q

xlz, 2017-05-07:
USB problem. You didn't receive any depth data.

Start here https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#linux and see if anything helps, otherwise make a proper bug report.

blueshark000, 2017-05-08:
Thank you very much,I have solved this question. @xlz

closed 2017-05-08 by xlz

leviresende, 2019-05-14:
Hey @xlz 
I'm having the same issue. I'm trying to connect a regular TX2 with kinect v2 and I'm heading some troubles...
Now my current log error is:

**$./bin/Protonect**
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 194355135047
[Info] [Freenect2Impl] found 1 devices
[Info] [CudaDepthPacketProcessorImpl] device 0: NVIDIA Tegra X2 @ 1300MHz Memory 7846MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 194355135047
device firmware: 4.0.3916.0
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [DepthPacketStreamParser] 1 packets were lost
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [CudaDepthPacketProcessor] avg. time: 4.67102ms -> ~214.086Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 33.6431ms -> ~29.7238Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 4.34132ms -> ~230.344Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 28.2427ms -> ~35.4074Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 4.32481ms -> ~231.224Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 27.9939ms -> ~35.7221Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 4.39239ms -> ~227.666Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
^C[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[... 142 lines cut ...]

In addition my configurations are:
L4T 28.2.1 [ JetPack 3.3 or 3.2.1 ]
Ubuntu 16.04.6 LTS
Kernel Version: 4.4.38-tegra
CUDA 9.0.252

And, just for you know, I've needed to change the file cmakelists.txt as follow to be possible compile libfreenect2:

SET(MY_DIR ${libfreenect2_SOURCE_DIR})
SET(DEPENDS_DIR "${MY_DIR}/depends" CACHE STRING "dependency directory must be set to 'false' if external deps are used")

OPTION(BUILD_SHARED_LIBS "Build shared (ON) or static (OFF) libraries" ON)
OPTION(BUILD_EXAMPLES "Build examples" ON)
OPTION(BUILD_OPENNI2_DRIVER "Build OpenNI2 driver" ON)
OPTION(ENABLE_CXX11 "Enable C++11 support" ON)
OPTION(ENABLE_OPENCL "Enable OpenCL support" OFF)
OPTION(ENABLE_CUDA "Enable CUDA support" ON)
OPTION(ENABLE_OPENGL "Enable OpenGL support" OFF)
OPTION(ENABLE_VAAPI "Enable VA-API support" OFF)
OPTION(ENABLE_TEGRAJPEG "Enable Tegra HW JPEG support" OFF)
OPTION(ENABLE_PROFILING "Collect profiling stats (memory consuming)" OFF)

Do you have any clue what can I to do now?
Thanks in advance.