OpenKinect/libfreenect2 #842: protonect.exe stop to work

chengzijiazu, 2017-05-07:
Overview Description:
Hi,everyone
For the past seveal days,I was trying to build the libfreenect2 for my Multiple-Kinect project. But I got the problem that protonect.exe stop to work!!! I retry it for many times, but it get the same issues. Could anyone give a hand?Thinks a lot! 

Environment variables: LOGFILE=<protonect.log>
Usage: C:\libfreenect2-master\build\bin\Debug\Protonect.exe [-gpu=<id>] [gl | cl
 | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected

Version, Platform, and Hardware Bug Found:
Version: 0.2.0
PlatForm:Windows8.1 x64

Now, when I download the libfreenect2-0.2.0-libusbk-vs2015-x64.zip and It work, but when I build it in the VS2015, problem came out. What cause this problem,how could i fix?

xlz, 2017-05-07:
If you have this exception at least you could let it enter the visual studio debugger and show us the line where the exception happened.

chengzijiazu, 2017-05-08:
@xlz Thinks your reply! The visual studio 2015 debugger show the problem code line 
`	p->apib = &usb_api_backend[USB_API_UNSUPPORTED];
	p->sub_api = SUB_API_NOTSET;
	for (i = 0; i < USB_MAXINTERFACES; i++) {
		p->usb_interface[i].apib = &usb_api_backend[USB_API_UNSUPPORTED];
		p->usb_interface[i].sub_api = SUB_API_NOTSET;
	}

	return p;
}

static inline void windows_device_priv_release(struct libusb_device *dev)
{
	struct windows_device_priv *p = _device_priv(dev);
	int i;

	free(p->path);
	for (i = 0; i < dev->num_configurations; i++)
		free(p->config_descriptor[i]); //This is the problem code which the debugger show
	free(p->config_descriptor);
	free(p->hid);
	for (i = 0; i < USB_MAXINTERFACES; i++) {
		free(p->usb_interface[i].path);
		free(p->usb_interface[i].endpoint);
	}
}

struct interface_handle_t {
	HANDLE dev_handle; // WinUSB needs an extra handle for the file
	HANDLE api_handle; // used by the API to communicate with the device
};

struct windows_device_handle_priv {
	int active_interface;`
This the screen picture: https://pan.baidu.com/s/1hsaaEJM

xlz, 2017-05-08:
What is the stack trace? I only see the last function `windows_device_priv_release`.

chengzijiazu, 2017-05-08:
the stack trace:
C4996	'std::copy::_Unchecked_iterators::_Deprecate': Call to 'std::copy' with parameters that may be unsafe - this call relies on the caller to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNINGS. See documentation on how to use Visual C++ 'Checked Iterators'	freenect2	C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\xutility	2372

xlz, 2017-05-08:
This is not the stack trace.

chengzijiazu, 2017-05-08:
https://pan.baidu.com/s/1gfqrJ5h

chengzijiazu, 2017-05-08:
[https://pan.baidu.com/s/1gfqrJ5h](url)

xlz, 2017-05-08:
Please rerun it with the environment variable `LIBUSB_DEBUG=4` and redirect the output to a log file and post it here.

chengzijiazu, 2017-05-08:
yes, But I may have to learn how to set the environment variable and redirect the output log file. That is not my fault...

chengzijiazu, 2017-05-08:
@xlz I rebuild the solution, and some warning message show that:
`
“C:\libfreenect2\depends\libusb_src\msvc\libusb_usbdk_dll_2015.vcxproj”(Rebuild 目标) (1) ->
(Link 目标) ->
  libusb-usbdk-1.0.exp : warning LNK4070: .EXP 中的 /OUT:libusb-1.0.dll 指令与输出文件名“C:\libfreenect2\depends\libusb_
src\msvc\
..\x64\Release\dll\libusb-usbdk-1.0.dll”不同；忽略指令 [C:\libfreenect2\depends\libusb_src\msvc\libusb_usbdk_dll_2015.vc
xproj]
  libusb-usbdk-1.0.exp : warning LNK4070: .EXP 中的 /OUT:libusb-1.0.dll 指令与输出文件名“C:\libfreenect2\depends\libusb_
src\msvc\
..\x64\Release\dll\libusb-usbdk-1.0.dll”不同；忽略指令 [C:\libfreenect2\depends\libusb_src\msvc\libusb_usbdk_dll_2015.vc
xproj]

    2 个警告
    0 个错误
`

xlz, 2017-05-09:
I will need the log file to determine the problem or I will wait until some more time before I can get a Windows machine to test this.

xlz, 2017-05-11:
I believe this is a regression in libusb. See related report https://github.com/libusb/libusb/issues/300.

Only libusbK users are affected. You can use usbdk as a workaround atm (it's the recommended default anyway).

xlz, 2017-05-12:
Fixed in upstream.

closed 2017-05-12 by xlz