OpenKinect/libfreenect2 #844: not all subsequences received / packets were lost by using multiple kinects

dimitar-rakov, 2017-05-08:
Hi,

I am trying to work with two Kinect  sensors on Ubuntu14.04, but there is **[DepthPacketStreamParser] not all subsequences received** and **[DepthPacketStreamParser]  packets were lost** messages. I am using  CUDA 7.5 and I set already  USBFS buffer size to 256. I tried also with **-noviewer** and **-norgb** option, which does not seem to have a serious impact.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline` `e900ef1`
2. `uname -a` 
`4.4.0-75-generic #96~14.04.1-Ubuntu SMP Thu Apr 20 11:06:30 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
`
3. `lsusb -t`
`/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 4: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 7: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 9: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 10: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
`
4. `lspci -nn`
`00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:02.0 Display controller [0380]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller [8086:0412] (rev 06)
00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
00:1c.1 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 [8086:8c12] (rev d5)
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
00:1c.4 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 [8086:8c18] (rev d5)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation C220 Series Chipset Family H81 Express LPC Controller [8086:8c5c] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 750] [10de:1381] (rev a2)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:0fbc] (rev a1)
03:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller [1912:0014] (rev 03)
04:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)
05:00.0 USB controller [0c03]: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller [1b21:1142]
`
Steps to Reproduce:

1. `./bin/Protonect cuda  -noviewer -norgb 011296264147`
2. `./bin/Protonect cuda -noviewer -norgb 003722754847`

Actual Results:
without LIBUSB_DEBUG
[log-Protonect cuda  -noviewer -norgb 011296264147.txt](https://github.com/OpenKinect/libfreenect2/files/983956/log-Protonect.cuda.-noviewer.-norgb.011296264147.txt)
[log-Protonect cuda -noviewer -norgb 003722754847.txt](https://github.com/OpenKinect/libfreenect2/files/983957/log-Protonect.cuda.-noviewer.-norgb.003722754847.txt)

with LIBUSB_DEBUG=4. Files are quite large.
[log - LIBUSB_DEBUG=4  Protonect cuda  -noviewer -norgb 011296264147.txt](https://github.com/OpenKinect/libfreenect2/files/984034/log.-.LIBUSB_DEBUG.4.Protonect.cuda.-noviewer.-norgb.011296264147.txt)
[log - LIBUSB_DEBUG=4 Protonect cuda -noviewer -norgb 003722754847.txt](https://github.com/OpenKinect/libfreenect2/files/984035/log.-.LIBUSB_DEBUG.4.Protonect.cuda.-noviewer.-norgb.003722754847.txt)

I am adding also an individual log from `dmesg`

[log- dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/984064/log-.dmesg.txt)

Any help would be really appreciated.

xlz, 2017-05-08:
Having a few packet losses is quite normal.

dimitar-rakov, 2017-05-08:
I just made some quick tests to find out an approximate amount of dropped packages. It seems that even with rgb and viewer, the first usb interface (on board) drops 3-5 packages and the processing is around:
`[Info] [CudaDepthPacketProcessor] avg. time: 4.59248ms -> ~217.748Hz`
`[Info] [VaapiRgbPacketProcessor] avg. time: 6.90345ms -> ~144.855Hz`

However, the second interface (pcie extension card ) has always worser performance and it drops more than 30 packages quite often. The processing is something like: 
`[Info] [CudaDepthPacketProcessor] avg. time: 4.61196ms -> ~216.828Hz`
`[Info] [VaapiRgbPacketProcessor] avg. time: 9.1503ms -> ~109.286Hz`

How many dropped packages can be considered as normal? Could the higher drop rate for the second interface be due to the card?

xlz, 2017-05-08:
I think the previous logs you posted showed about 1% loss rate, which is pretty normal.

The card and its driver are definitely a factor. ASM1042A is pretty bad but I don't know if you were using it or the Renesas one. But the flood of messages in dmesg is definitely not a good sign for whichever the card you were using.

dimitar-rakov, 2017-05-09:
I am using the Renesas controller. The ASM1042A does not work at all. Do you have experience with some PCIe x1 extension card, which is known to work properly? I need only one additional usb controller, since I am not going to use more than two sensors on this PC.

xlz, 2017-05-11:
https://github.com/OpenKinect/libfreenect2#requirements-for-multiple-kinects
That's all we have.

Also make sure you use the newest kernel with improved USB 3 controller drivers.

closed 2017-05-11 by xlz