OpenKinect/libfreenect2 #845: Color_Depth_Map returned from registration.apply() is Frame::Invalid

Adamwgoh, 2017-05-09:
Steps to Reproduce:

1. Create Frame object, and use with Registration.apply() function.
2. Check Frame object.
3. Returns Frame::Invalid

SirDifferential, 2017-05-09:
Please include the relevant part of your code that produces this.

xlz, 2017-05-09:
What do you mean? Well, indeed `Registration::apply()` doesn't update `Frame::format` so if a `Frame` object is created with `Invalid` format it stays `Invalid`. Is this something a problem for you?

Adamwgoh, 2017-05-09:
sorry Header might be misleading, I was referring to big_depth. We passed in the Frame as constructed using this, but the return is still empty.

`Frame::Frame(size_t width, size_t height, size_t bytes_per_pixel, unsigned char *data_) `

Also to slot in a question here, how do I access the color_depth_map? Is it a uv coordinate map? If so, do we pass in a int* array of same width*height as the depth or color frame?

xlz, 2017-05-11:
What is "empty"? I don't think your issue is reproducible, even without you providing any code.

> how do I access the color_depth_map?

The docs said it very clearly:

> int * 	color_depth_map = 0 
> color_depth_map	Index of mapped color pixel for each depth pixel (512x424).

You pass a pointer to a `int[512*424]`, and each element in the array is an index of mapped color pixel for the corresponding depth pixel. Can't say it more literally.

Adamwgoh, 2017-05-26:
Hi @xlz My bad for not understanding it correctly. I managed to get it running. Thanks for your patience and understanding.

closed 2017-05-26 by Adamwgoh

Hansansui-cv, 2021-08-28:
@xlz hi, could you please tell me that why there is -1 in the color_map_depth? what is that mean? And, when i trying getting the mapped point in depth from color, why i can not get the index from color_depth_map? i need your help, thank you.