OpenKinect/libfreenect2 #846: package 'libusb-1.0>=1.0.20' not found

DaiongAI, 2017-05-09:
dpkg -l libusb-1.0-*:

ii  libusb-1.0-0:a 2:1.0.20-1pp amd64        userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.20-1pp amd64        userspace USB programming library
ii  libusb-1.0-doc 2:1.0.20-1pp all          documentation for userspace USB p

DaiongAI, 2017-05-09:
what should i do?

xlz, 2017-05-09:
`pkg-config --debug 'libusb-1.0 >= 1.0.20'; echo $?`
?

DaiongAI, 2017-05-10:
pkg-config --debug 'libusb-1.0 >= 1.0.20'; echo $?:
'''
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
  line>
Path position of 'libusb-1.0' is 4
Adding 'libusb-1.0' to list of known packages, returning as package 'libusb-1.0'
1
'''
but unuseful!

xlz, 2017-05-10:
Your libusb installation is bad.

This is what a normal libusb installation looks like
```
Reading 'libusb-1.0' from file '/usr/lib/x86_64-linux-gnu/pkgconfig/libusb-1.0.pc'
Parsing package file '/usr/lib/x86_64-linux-gnu/pkgconfig/libusb-1.0.pc'
  line>prefix=/usr
 Variable declaration, 'prefix' has value '/usr'
  line>exec_prefix=${prefix}
 Variable declaration, 'exec_prefix' has value '/usr'
  line>libdir=${prefix}/lib/x86_64-linux-gnu
 Variable declaration, 'libdir' has value '/usr/lib/x86_64-linux-gnu'
  line>includedir=${prefix}/include
 Variable declaration, 'includedir' has value '/usr/include'
  line>
  line>Name: libusb-1.0
  line>Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace
  line>Version: 1.0.21
  line>Libs: -L${libdir} -lusb-1.0
  line>Libs.private: -ludev  -pthread
Unknown keyword 'Libs.private' in '/usr/lib/x86_64-linux-gnu/pkgconfig/libusb-1.0.pc'
  line>Cflags: -I${includedir}/libusb-1.0
Path position of 'libusb-1.0' is 4
Adding 'libusb-1.0' to list of known packages
Package libusb-1.0 has -L /usr/lib/x86_64-linux-gnu in Libs
Removing -L /usr/lib/x86_64-linux-gnu from libs for libusb-1.0
0
```

closed 2017-05-10 by xlz