OpenKinect/libfreenect2 #848: question: [Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable

zhoumh1, 2017-05-10:
I did that and also in the libfreenect2/depends directory I run the install_libsub.sh because I understand that libfreenect2 needs its own version of libusb. I´m almost done. Now It´s founding the Kinect v2 but I have a message: [Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window
I have dealed with the problem for several days, can anyone help me? Thank you!

xlz, 2017-05-10:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#opengl-only-check-driver-version

closed 2017-05-10 by xlz

Booooooosh, 2017-06-26:
Did you resolve the problem finally? I am facing the same problem, it seems that the graphic driver version is not higher than 3.1, however, it is not specified how to fix this problem for Ubuntu 14.04