OpenKinect/libfreenect2 #849: Unit of timestamp

caselitz, 2017-05-11:
Hi,

I was wondering why replaying images based on timestamps was running a bit fast, so I quickly investigated this.

I think it was once discussed here: https://github.com/OpenKinect/libfreenect2/issues/413 - I assume this quote
>  it appears to be either roughly or exactly 1/10 of a millisecond

led to the documentation in frame_listener.hpp

> uint32_t timestamp;     ///< Unit: roughly or exactly 0.1 millisecond

I followed @xlz recommendation and measured against real-time by recording a stop watch. Turns out, the **factor on the uint32_t** is pretty exactly **1.25 * 1e-4**. The diff of the timestamps between two consecutive frames are reliably 266/267 or 533/534. Multiplication with the factor yields ~33ms or ~66ms.

The Kinect2 seems to force a frame rate of either 30fps or 15fps depending on lighting conditions. The auto-exposure also seems to take this into account - exposure time is max 66ms and then quickly jumps to values below 33ms to achieve the 30fps instead of 15fps. 

Not sure if this is known already, but I think it would be nice to update the documentation accordingly.

Best,
Tim

xlz, 2017-05-11:
Is this the case for both RGB and depth timestamps?

caselitz, 2017-05-11:
Hm, good question. I only checked for RGB and assumed that IR is same. But I will verify that...

xlz, 2017-05-11:
```
rgb seq, rgb ts, ir seq, ir ts
647 200284 85 200104
648 200551 86 200371
649 200817 87 200638
```
OK, it looks like IR is the case too. Actually this is also the bus interval of a iso microframe (125us). Perhaps the onboard timestamps are derived from the USB clock.

But the temporal synchronization between RGB and IR looks quite problematic right now.

commit 0fb41986 (2017-05-11): docs: Update timestamp unit

Thanks to @caselitz's testing

Close #849.
closed 2017-05-11 by xlz