OpenKinect/libfreenect2 #850: How to integrate the libfreenect2 to the YOLO detection system?

Abbyls, 2017-05-14:
Hello,
       I want to use the yolo to detecting the objects on TX1 from the kinect v2 camera, but i don't know how to get the rgb stream which get by 
` libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];//BGRA format`  in libfreenct2.
but the YOLO get the video stream by 
`cap = cvCaptureFromCAM(cam_index);` in darknet.
my question is how to get the rgb stream by cvCaptureFromCAM
 when i opened the libfreenect2 driver, and then run ./darknet detector demo cfg/voc.data cfg/tiny-yolo-voc.cfg tiny-yolo-voc.weights
       the error is :
           Couldn't connect to webcam: No such file or directory
      Is there anyone know how to solve it?

floe, 2017-05-14:
If you just want to run the YOLO detector, then why do you need the Kinect v2 at all? If you're just interested in the RGB camera stream, a plain old webcam will be a much better choice and will work with OpenCV out of the box.

Abbyls, 2017-05-14:
I want to localize the object's 3D position in real time, so, i think i can use the rgb camera stream to get the position from rgb image, and combined with the depth information to get the 3D position.

floe, 2017-05-14:
Ah, that sounds sensible. libfreenect2 doesn't support the OpenCV capture API, so after you have the `Frame` object, you need to convert it to a `cv::Mat` yourself and then pass that to the YOLO detector. Here's an example: https://github.com/OpenKinect/libfreenect2/issues/828#issuecomment-294307867 (note that this is for the depth frame, not RGB).

Abbyls, 2017-05-14:
OK, thank you very much!

Abduoit, 2017-07-06:
Hi, 

>  @Abbyls

I am working in the same pool, depth information by using kinect v2 and YOLO darknet.

would you please tell me how did you localize the object's 3D position in real time by using kinect v2. ??

I really appreciate your time

Abbyls, 2017-07-15:
Hi, @Abduoit 
Thank you for your interest in the same problem.
1. I resized the rgb image(1920*1080) to the size of depth imag(512*424)  and get the depth image's pointcloud.
2. I transform the resized rgb data to the yolo to get the 2D position in the rgb image, and then, combined the depth pointcloud to find the corresponding 3D position.

Abduoit, 2017-07-15:
Thanks @Abbyls for reply 

Are there any resources (links) with more details, to make it easier.

Abduoit, 2017-07-18:
Hi, @Abbyls 

I found this [link](https://github.com/chihyaoma/KinectOneStream) it might help you to convert streaming images into `cv::mat` in OpenCV. It is for depth and rgb images

 Since I am new in OpenCv, and libfreenect2, I actually couldn't use it correctly. If you use it correctly please let me know,

closed 2017-12-01 by xlz