OpenKinect/libfreenect2 #851: Memory Leak with Consecutive Calls to "waitForNewFrame(...)"

wired365, 2017-05-16:
I'm not sure if this was a design intent or an actual bug but there appears to be a memory leak when `release()` is not called in between any function that gets a new frame, such as `waitForNewFrame(FrameMap &frame)`. Looking at the source code it appears that the `FrameMap` reference that is passed into the function is overwritten by the newly received `FrameMap` leaving the old `FrameMap` data allocated without any reference to it.

It seems like a pretty easy situation to end up in, but that could just be me. Whether it is a bug or not I wanted others to at least be aware that they have to either call `release(FrameMap &frame)` once they are finished with that current frame or must implement some form of memory control like manually deleting or smart pointers to the frame data.

HenningJ, 2017-05-16:
I think the [documentation](https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1SyncMultiFrameListener.html#abe370825e64b1cca85f7e7be8bd986bf) of `waitForNewFrame(FrameMap &frame)` is pretty clear about that:

    [out]	frame	Caller is responsible to release the frames.

But if you can think of a way to make it more clear...

wired365, 2017-05-16:
Yeah I had caught that, and it makes a lot more sense given what I know now. The part that threw me was the documentation for `release(FrameMap & frame)`:

`Shortcut to delete all frames.`

I had incorrectly interpreted that to mean deletes all non-released frames that have been collected. I didn't realize it meant the "frames" in the "FrameMap". That was a mistake on my part. Until I looked at the source code I had naively assumed that either the listener or device was keeping track of the `FrameMap` references. Maybe change the definition of release to explicitly state it only releases the `FrameMap` that is passed to it?

commit 0856a20e (2017-12-01): docs: Clarify SyncMultiFrameListener API usage

Fix #851.
closed 2017-12-01 by xlz