OpenKinect/libfreenect2 #853: No rule to make target `/usr/local/lib/libglfw3.a', needed by `lib/libfreenect2.so.0.2.0'

fedpet, 2017-05-17:
Hello, I'm having a little issue trying to compile due to my limited experience with make and cmake

Version, Platform, and Hardware:
git log -1 --oneline = 6236635 docs: Fix a typo
uname -a = Linux 3.13.0-39-generic #66-Ubuntu SMP Tue Oct 28 13:30:27 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

output of make:
[ 88%] Built target generate_resources_tool
make[2]: *** No rule to make target `/usr/local/lib/libglfw3.a', needed by `lib/libfreenect2.so.0.2.0'.  Stop.
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

thanks for your help

floe, 2017-05-17:
What does `pkg-config --libs glfw3` show?

fedpet, 2017-05-18:
`-lglfw`

floe, 2017-05-18:
OK, so it looks like your system has GLFW installed correctly, but there's a second installation in /usr/local/lib (generally not such a good idea). What does `ls -l /usr/local/lib` show?

fedpet, 2017-05-18:
```
total 769044
drwxr-xr-x 3 root root       4096 mar 30 17:58 cmake
lrwxrwxrwx 1 root root         22 mar 13 11:54 libopencv_aruco.so -> libopencv_aruco.so.3.2
lrwxrwxrwx 1 root root         24 mar 13 11:54 libopencv_aruco.so.3.2 -> libopencv_aruco.so.3.2.0
-rw-r--r-- 1 root root     361587 apr  6 21:31 libopencv_aruco.so.3.2.0
lrwxrwxrwx 1 root root         23 mar 13 11:54 libopencv_bgsegm.so -> libopencv_bgsegm.so.3.2
lrwxrwxrwx 1 root root         25 mar 13 11:54 libopencv_bgsegm.so.3.2 -> libopencv_bgsegm.so.3.2.0
-rw-r--r-- 1 root root      74456 apr  6 21:31 libopencv_bgsegm.so.3.2.0
lrwxrwxrwx 1 root root         28 mar 13 11:54 libopencv_bioinspired.so -> libopencv_bioinspired.so.3.2
lrwxrwxrwx 1 root root         30 mar 13 11:54 libopencv_bioinspired.so.3.2 -> libopencv_bioinspired.so.3.2.0
-rw-r--r-- 1 root root     263025 apr  6 21:31 libopencv_bioinspired.so.3.2.0
lrwxrwxrwx 1 root root         24 mar 13 11:54 libopencv_calib3d.so -> libopencv_calib3d.so.3.2
lrwxrwxrwx 1 root root         26 mar 13 11:54 libopencv_calib3d.so.3.2 -> libopencv_calib3d.so.3.2.0
-rw-r--r-- 1 root root    1741023 apr  6 21:31 libopencv_calib3d.so.3.2.0
lrwxrwxrwx 1 root root         23 mar 13 11:54 libopencv_ccalib.so -> libopencv_ccalib.so.3.2
lrwxrwxrwx 1 root root         25 mar 13 11:54 libopencv_ccalib.so.3.2 -> libopencv_ccalib.so.3.2.0
-rw-r--r-- 1 root root     514984 apr  6 21:31 libopencv_ccalib.so.3.2.0
lrwxrwxrwx 1 root root         21 mar 13 11:54 libopencv_core.so -> libopencv_core.so.3.2
lrwxrwxrwx 1 root root         23 mar 13 11:54 libopencv_core.so.3.2 -> libopencv_core.so.3.2.0
-rw-r--r-- 1 root root   13117959 apr  6 21:31 libopencv_core.so.3.2.0
lrwxrwxrwx 1 root root         27 mar 22 11:53 libopencv_cudaarithm.so -> libopencv_cudaarithm.so.3.2
lrwxrwxrwx 1 root root         29 mar 22 11:53 libopencv_cudaarithm.so.3.2 -> libopencv_cudaarithm.so.3.2.0
-rw-r--r-- 1 root root  148927022 mar 22 11:37 libopencv_cudaarithm.so.3.2.0
lrwxrwxrwx 1 root root         27 mar 22 11:53 libopencv_cudabgsegm.so -> libopencv_cudabgsegm.so.3.2
lrwxrwxrwx 1 root root         29 mar 22 11:53 libopencv_cudabgsegm.so.3.2 -> libopencv_cudabgsegm.so.3.2.0
-rw-r--r-- 1 root root    1433143 mar 22 11:38 libopencv_cudabgsegm.so.3.2.0
lrwxrwxrwx 1 root root         26 mar 22 11:53 libopencv_cudacodec.so -> libopencv_cudacodec.so.3.2
lrwxrwxrwx 1 root root         28 mar 22 11:53 libopencv_cudacodec.so.3.2 -> libopencv_cudacodec.so.3.2.0
-rw-r--r-- 1 root root     179182 mar 22 11:41 libopencv_cudacodec.so.3.2.0
lrwxrwxrwx 1 root root         31 mar 22 11:53 libopencv_cudafeatures2d.so -> libopencv_cudafeatures2d.so.3.2
lrwxrwxrwx 1 root root         33 mar 22 11:53 libopencv_cudafeatures2d.so.3.2 -> libopencv_cudafeatures2d.so.3.2.0
-rw-r--r-- 1 root root   27982030 mar 22 11:43 libopencv_cudafeatures2d.so.3.2.0
lrwxrwxrwx 1 root root         28 mar 22 11:53 libopencv_cudafilters.so -> libopencv_cudafilters.so.3.2
lrwxrwxrwx 1 root root         30 mar 22 11:53 libopencv_cudafilters.so.3.2 -> libopencv_cudafilters.so.3.2.0
-rw-r--r-- 1 root root  392845625 mar 22 11:39 libopencv_cudafilters.so.3.2.0
lrwxrwxrwx 1 root root         28 mar 22 11:53 libopencv_cudaimgproc.so -> libopencv_cudaimgproc.so.3.2
lrwxrwxrwx 1 root root         30 mar 22 11:53 libopencv_cudaimgproc.so.3.2 -> libopencv_cudaimgproc.so.3.2.0
-rw-r--r-- 1 root root   29006063 mar 22 11:39 libopencv_cudaimgproc.so.3.2.0
lrwxrwxrwx 1 root root         27 mar 22 11:53 libopencv_cudalegacy.so -> libopencv_cudalegacy.so.3.2
lrwxrwxrwx 1 root root         29 mar 22 11:53 libopencv_cudalegacy.so.3.2 -> libopencv_cudalegacy.so.3.2.0
-rw-r--r-- 1 root root    8085300 mar 22 11:43 libopencv_cudalegacy.so.3.2.0
lrwxrwxrwx 1 root root         30 mar 22 11:53 libopencv_cudaobjdetect.so -> libopencv_cudaobjdetect.so.3.2
lrwxrwxrwx 1 root root         32 mar 22 11:53 libopencv_cudaobjdetect.so.3.2 -> libopencv_cudaobjdetect.so.3.2.0
-rw-r--r-- 1 root root    1149077 mar 22 11:43 libopencv_cudaobjdetect.so.3.2.0
lrwxrwxrwx 1 root root         28 mar 22 11:53 libopencv_cudaoptflow.so -> libopencv_cudaoptflow.so.3.2
lrwxrwxrwx 1 root root         30 mar 22 11:53 libopencv_cudaoptflow.so.3.2 -> libopencv_cudaoptflow.so.3.2.0
-rw-r--r-- 1 root root   52158973 mar 22 11:43 libopencv_cudaoptflow.so.3.2.0
lrwxrwxrwx 1 root root         27 mar 22 11:53 libopencv_cudastereo.so -> libopencv_cudastereo.so.3.2
lrwxrwxrwx 1 root root         29 mar 22 11:53 libopencv_cudastereo.so.3.2 -> libopencv_cudastereo.so.3.2.0
-rw-r--r-- 1 root root   10297459 mar 22 11:43 libopencv_cudastereo.so.3.2.0
lrwxrwxrwx 1 root root         28 mar 22 11:53 libopencv_cudawarping.so -> libopencv_cudawarping.so.3.2
lrwxrwxrwx 1 root root         30 mar 22 11:53 libopencv_cudawarping.so.3.2 -> libopencv_cudawarping.so.3.2.0
-rw-r--r-- 1 root root   52023817 mar 22 11:39 libopencv_cudawarping.so.3.2.0
lrwxrwxrwx 1 root root         22 mar 22 11:53 libopencv_cudev.so -> libopencv_cudev.so.3.2
lrwxrwxrwx 1 root root         24 mar 22 11:53 libopencv_cudev.so.3.2 -> libopencv_cudev.so.3.2.0
-rw-r--r-- 1 root root       7663 mar 21 21:56 libopencv_cudev.so.3.2.0
lrwxrwxrwx 1 root root         25 mar 13 11:54 libopencv_datasets.so -> libopencv_datasets.so.3.2
lrwxrwxrwx 1 root root         27 mar 13 11:54 libopencv_datasets.so.3.2 -> libopencv_datasets.so.3.2.0
-rw-r--r-- 1 root root     680235 apr  6 21:31 libopencv_datasets.so.3.2.0
[... cut ...]
-rw-r--r-- 1 root root     120464 apr  6 21:31 libopencv_xobjdetect.so.3.2.0
lrwxrwxrwx 1 root root         23 mar 13 11:54 libopencv_xphoto.so -> libopencv_xphoto.so.3.2
lrwxrwxrwx 1 root root         25 mar 13 11:54 libopencv_xphoto.so.3.2 -> libopencv_xphoto.so.3.2.0
-rw-r--r-- 1 root root     528140 apr  6 21:31 libopencv_xphoto.so.3.2.0
drwxr-xr-x 2 root root       4096 mag 17 16:09 pkgconfig
drwxrwsr-x 4 root staff      4096 mar 13 10:42 python2.7
drwxrwsr-x 3 root staff      4096 nov  5  2014 python3.4****
```

floe, 2017-05-18:
Hm. There's no libglfw3.a, so I'm starting to think that you might just need to wipe your build directory and re-run cmake.

xlz, 2017-05-19:
`make VERBOSE=1`

This is weird. There a plenty of bug reports about the static library `libglfw3.a`. I don't know where this thing comes from. These issues are never reproducible and after a while they seem to solve themselves.

fedpet, 2017-05-19:
ok figured things out (hopefully).
I think libglfw3 problems come from manually installing this http://www.glfw.org/

Then, during installation, i ran  `sudo apt-get install libgl1-mesa-dri-lts-vivid` which is a big NO NO. This pretty much destroyed my graphical system (Ubuntu Mate desktop 14.04) and had to reinstall `ubuntu-mate-core ubuntu-mate-desktop xorg xserver-xorg` but it was my fault for not being carefull enough.

After having recovered my system I trashed the whole libfreenect2 directory, re-downloaded everything and followed the installation tutorial again (without installing libgl1-mesa-dri-lts-vivid)
The build ran succesfully!

Now i have another problem but I don't know if it's related. I get this output running Protonect: https://pastebin.com/vpJqTriy
I think theese are the main errors:
```
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

failed to link shader program!
error: linking with uncompiled shadererror: linking with uncompiled shader
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

failed to link shader program!
```

Thank you very much for your help and support

floe, 2017-05-19:
It looks like this is actually a common error with GLFW, and it is caused by wrong library ordering on the gcc commandline: https://stackoverflow.com/questions/17559280/how-do-i-build-my-glfw3-project-on-linux

floe, 2017-05-19:
Regarding that vertex shader error, the problem is exactly what the error message says: your GPU doesn't support GLSL 3.30. Depending on what GPU you have, this may be fixable with a more recent/proprietary driver, but quite possibly you need a newer GPU. You can try `protonect cpu` and see if that works, but it will be very slow.