OpenKinect/libfreenect2 #854: How to adjust RGB-Depth registration performed by the driver?

mtee, 2017-05-18:
I have compiled all the dependencies and the driver according to the instructions and everything runs fine, however the 'registered" depth and RGB are [image]

The same obviously also happens in all other programs that rely on registered frames from libfreenect2.
It is to expect that the factory calibration will not be perfect, but the registration seems to be off by just a few pixels in one direction. How do I proceed about fixing it? Is it possible without driver recompilation?

It might be a completely isolated issue, but the openni2 wrapper registration with the libfreenect2 is even worse than that:
[image]

floe, 2017-05-18:
I can't tell what the problem with the openni2 wrapper is right now.
That being said, the original registration looks just as expected - the "misalignment" you're seeing is simply the shadow your hand is casting in infrared (i.e. no depth data is available there).

mtee, 2017-05-18:
The shadow is not the issue. A part of the hand has been registered to the depth pixels of the background. I have made a short video to better illustrate the problem:
http://gph.is/2qv3m2v
as you see, quite a few pixels of the contour of the red chair have been mapped to the wrong depths - those of the background wall.

floe, 2017-05-18:
OK, misunderstood your question. I think what you're seeing is more or less the expected error for the factory calibration, it's only "good enough". If you perform your own calibration, e.g. with OpenCV and a chessboard pattern, then you will probably get better results.

closed 2018-11-18 by xlz