OpenKinect/libfreenect2 #855: Multiple kinects using threads

sairamkolla, 2017-05-19:
Hi,

I've run 3 kinects simultaneously on different instances of protonect and it's working perfectly fine. But when I try to run them using multiple threads in same program, I get segmentation fault. Any idea why this is happening?

Regards,
sairam

xlz, 2017-05-20:
What's the code? Where's the segfault?

sairamkolla, 2017-05-20:
I'm getting segfault at this line

`dev = freenect2.openDevice(serial);`

Also, the code is working perfectly well if I run only one thread. Please let me know if you want to look at the code.

Regards,
sairam

sairamkolla, 2017-05-24:
I've set the environmental variables as shown [here](https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#multiple-kinects),but I get the same segmentation fault.

Also, when I  run multiple instances  of protonect  in different threads,it's working completely fine

`int status = system("~/libfreenect2/build/bin/Protonect serial")`

floe, 2017-05-24:
No way to give any feedback without actually seeing your code...

sairamkolla, 2017-05-24:
Thanks for the reply @floe 

Here are my  [code](https://pastebin.com/uHax8j5J) and [CMakelists.txt](https://pastebin.com/nChW3S8N).

Regards
sairam

floe, 2017-05-24:
Two things:

- You never call `enumerateDevices()` (see https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Freenect2.html#ac7c87454b2270b68ed326e4022f72997), which is probably the main cause for your crash.
- You have one `libfreenect2::Freenect2` object per thread, it would probably be better to have only a single global one and only pass pointers to `libfreenect2::Freenect2Device` to each thread.

sairamkolla, 2017-05-25:
It's working now. thanks!

closed 2017-05-25 by sairamkolla