OpenKinect/libfreenect2 #856: Protonect crashes when run in termincal

sagniknitr, 2017-05-21:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. Install all libfreenect2 successfully
2. run ./protonect
3. 

Actual Results: program crashes

Expected Results: shows kinect data

Reproducibility:

Additional Information:
[image]

xlz, 2017-05-21:
Most likely your hardware doesn't support Kinect2. https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#linux

sagniknitr, 2017-06-20:
Libfreenect2 is having some issues with AMD graphics drivers in Ubuntu. Twice I have installed it carefully following every step mentioned( including kernel update) , and every time the protonect executable runs only once after which it crashes. Also on restarting the laptop after installing and running protonect, the  Ubuntu xserver fails to start anytime after this. Only solution after this is to delete all my  AMD graphics driver from kernel mode.

I fail to understand why this is happening as I know correctly installed all my graphics driver ( all my OpenGL/OpenCL programs run without any issues). 

P.S. My graphics card is AMD Radeon 8730M

xlz, 2017-06-20:
Well, AMD drivers are known to be flaky. You can always run OpenCL using Intel beignet.

closed 2017-10-13 by floe