OpenKinect/libfreenect2 #857: Kinect v2 dont show anything but black screen

ektwr, 2017-05-28:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
6236635 docs: Fix a typo
2. `uname -a`
Linux john-P67A-D3-B3 3.13.0-24-generic #46-Ubuntu SMP Thu Apr 10 19:11:08 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`
john@john-P67A-D3-B3:~/libfreenect2/build$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 5, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 5, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 5: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 6: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M

4. `lspci -nn`
john@john-P67A-D3-B3:~/libfreenect2/build$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 2nd Generation Core Processor Family DRAM Controller [8086:0100] (rev 09)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200/2nd Generation Core Processor Family PCI Express Root Port [8086:0101] (rev 09)
00:16.0 Communication controller [0780]: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 [8086:1c3a] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 [8086:1c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller [8086:1c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 [8086:1c10] (rev b5)
00:1c.2 PCI bridge [0604]: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 3 [8086:1c14] (rev b5)
00:1c.3 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev b5)
00:1c.4 PCI bridge [0604]: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 5 [8086:1c18] (rev b5)
00:1d.0 USB controller [0c03]: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 [8086:1c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation P67 Express Chipset Family LPC Controller [8086:1c46] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 6 Series/C200 Series Chipset Family SATA AHCI Controller [8086:1c02] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller [8086:1c22] (rev 05)
01:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller [1912:0015] (rev 02)
03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 06)
04:00.0 PCI bridge [0604]: Integrated Technology Express, Inc. Device [1283:8892] (rev 30)
05:01.0 FireWire (IEEE 1394) [0c00]: VIA Technologies, Inc. VT6306/7/8 [Fire II(M)] IEEE 1394 OHCI Controller [1106:3044] (rev 46)
06:00.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8647 48-Lane, 3-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8647] (rev bb)
07:04.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8647 48-Lane, 3-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8647] (rev bb)
07:08.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8647 48-Lane, 3-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8647] (rev bb)
08:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Hemlock [Radeon HD 5970] [1002:689c]
08:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Cypress HDMI Audio [Radeon HD 5800 Series] [1002:aa50]
09:00.0 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] Hemlock [Radeon HD 5970] [1002:689c]

Steps to Reproduce:
1. 
sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/
./bin/Protonect 
2. 
3. 

Actual Results:
john@john-P67A-D3-B3:~/libfreenect2/build$ ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:7 with serial 001110441247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] started
device serial: 001110441247
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.1758ms -> ~43.1484Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 22.6671ms -> ~44.1168Hz

Then i have a black screen and freeze there

Expected Results:

Reproducibility:

Additional Information:

xlz, 2017-05-29:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#check-dmesg-after-running-protonect

ektwr, 2017-05-30:
Thank you for reply. I will try my kinect v2 device to a friend MSI  Z97 Gaming motherboard with PCI-e x16 gen2 with win 8.1 and put there the USB3 card i have to check my device there. I hope this way to realize if my kinect work to a more powerfull desktop than mine. I will be back with my results afterwords.

ektwr, 2017-06-02:
Good news for me!
Finaly i put my PCI-e card into a slot X16 to my friend computer and i found the problem.
I had forgot that when i buy the kinect v2, i did have the USB3 MS kit and i cut the cable to make a USB3 hack. As i didn't know if the device was working or not i didn't examine the insulation of the opened wires.
So after a couple of hours search and experiments i see that it was a small wire bridge between two cables who force the device to open and close imidiately.
Now i fix the cable and my device is working with my PC, not very speedy but works anyway.
I put the libfreenect2 and now i have a fully working kinect v2 device.
I 'm in touch with you because i will need advices for tracking examples too.
TIA

closed 2017-10-13 by floe