OpenKinect/libfreenect2 #858: Odroid XU4 problem.

hamster1979, 2017-05-30:
I'm trying to run ./Protonect -cl on regular Xbox One Kinect Sensor Bar along with a Kinect Adapter for Windows (because Kinect v2 for windows discontinued now) on Odroid XU4 platform and have this results:

`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
`[Debug] [DepthPacketStreamParser] subpacket too large`
Help please investigate this problem.
[uname.txt](https://github.com/OpenKinect/libfreenect2/files/1037793/uname.txt)
[clinfo.txt](https://github.com/OpenKinect/libfreenect2/files/1037764/clinfo.txt)
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/1037763/dmesg.txt)
[lsusb.txt](https://github.com/OpenKinect/libfreenect2/files/1037762/lsusb.txt)
[protonect_out.txt](https://github.com/OpenKinect/libfreenect2/files/1037761/protonect_out.txt)
[protonect_out_libusb_debug_4.txt](https://github.com/OpenKinect/libfreenect2/files/1037779/protonect_out_libusb_debug_4.txt)

xlz, 2017-06-01:
```
[ 1.855032] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 1.855051] [0000060d] libusb: debug [handle_iso_completion] overflow error
[ 1.866369] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 1.866385] [0000060d] libusb: debug [handle_iso_completion] overflow error
[ 1.871861] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 1.873059] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 1.873078] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 2.009487] [0000060d] libusb: debug [handle_iso_completion] low-level USB error -71
[ 2.010541] [0000060d] libusb: debug [handle_iso_completion] overflow error
[ 2.052488] [0000060d] libusb: debug [handle_iso_completion] overflow error
[ 2.052504] [0000060d] libusb: debug [handle_iso_completion] overflow error
[ 2.131213] [0000060d] libusb: debug [handle_bulk_completion] low level error -71
[ 2.131351] [0000060d] libusb: debug [handle_bulk_completion] low level error -71
[ 2.131516] [0000060d] libusb: debug [handle_bulk_completion] low level error -71
[ 2.131653] [0000060d] libusb: debug [handle_bulk_completion] low level error -71
```

Your dmesg is incomplete. Provide the dmesg after running Protonect.

hamster1979, 2017-06-02:
[dmesg_new.txt](https://github.com/OpenKinect/libfreenect2/files/1047152/dmesg_new.txt)
[protonect_out_libusb_debug_4_new.txt](https://github.com/OpenKinect/libfreenect2/files/1047161/protonect_out_libusb_debug_4_new.txt)

xlz, 2017-06-05:
No new information. The best I can guess is kernel problems. Try different kernels.

EOVERFLOW or EPROTO errors were generated from xhci stack.

hamster1979, 2017-06-05:
I tried now on newest odroid kernel 4.9.11 (https://forum.odroid.com/viewtopic.php?f=95&t=26986) - result the same. (( Any proposal?

closed 2018-11-18 by xlz

eric-schleicher, 2018-11-26:
@xlz does this mean that we'll never be able to run libfreenect2 on XU4? :(

xlz, 2018-11-26:
If I can get my hands on an actual XU4...

eric-schleicher, 2018-12-20:
let me see if i can get one to you...  Will email.