OpenKinect/libfreenect2 #863: it can't receive data on Mac Pro ?

rain3k, 2017-06-13:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
6236635 docs: Fix a typo
2. `uname -a`
Darwin kims-Mac-Pro.local 16.6.0 Darwin Kernel Version 16.6.0: Fri Apr 14 16:21:16 PDT 2017; root:xnu-3789.60.24~6/RELEASE_X86_64 x86_64
3. `lsusb -t`
  Bus 000.Dev 001: USB 2.0 Bus, 
  Bus 000.Dev 001: USB 3.0 Bus, 
    |__ Bus 000.Dev 012: iPhone, 480M
    |__ Bus 000.Dev 028: NuiSensor Adaptor, 480M
    |__ Bus 000.Dev 026: NuiSensor Adaptor, 480M
    |__ Bus 000.Dev 029: NuiSensor Adaptor, 5G
        |__ Bus 000.Dev 002: Xbox NUI Sensor, 5G
    |__ Bus 000.Dev 027: NuiSensor Adaptor, 5G
        |__ Bus 000.Dev 003: Xbox NUI Sensor, 5G
    |__ Bus 029.Dev 001: Hub, 480M
        |__ Bus 029.Dev 002: BRCM20702 Hub, 12M
            |__ Bus 029.Dev 005: Bluetooth USB Host Controller, 12M
4. `lspci -nn`

I have two computer (Mac Book Pro, Mac Pro)

I want to receive data from two Kinect on one machine. so I develop a  program which use libfreenect on Mac Book Pro(13-inch Retina Early 2015/macOS Sierra). 

but I want to use my Mac Pro. so..
copy my program to my Mac Pro, and run....

But... my program don't receive data from two Kinect on Mac Pro.

my program stuck this code (listener.waitForNewFrame(frames);)

it just receive data from one Kinect...

Plz Help me.

this is my code.

void kinectModule(libfreenect2::Freenect2Device *dev,int device_index)
{
    int types = 0;
    types |= libfreenect2::Frame::Color;
    types |= libfreenect2::Frame::Ir | libfreenect2::Frame::Depth;
    
    libfreenect2::SyncMultiFrameListener listener(types);
    libfreenect2::FrameMap frames;
    
    dev->setColorFrameListener(&listener);
    dev->setIrAndDepthFrameListener(&listener);
    
    if (dev->start())
    {
        std::cout << "device serial: " << dev->getSerialNumber() << std::endl;
        std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;
        
        while(!protonect_shutdown)
        {
//            std::cout << "kinectModule device_index"<< device_index << std::endl;
            
            listener.waitForNewFrame(frames);
            libfreenect2::Frame *rgbFrame = frames[libfreenect2::Frame::Color];
            libfreenect2::Frame *depthFrame = frames[libfreenect2::Frame::Depth];

            if(device_index == SERVERONE){
//                std::cout << "kinectModule One"<< std::endl;
                if(finishedOneTask){
                    finishedOneTask = false;
                    unsigned char* rgb_data = new unsigned char[rgbFrame->width * rgbFrame->height * rgbFrame->bytes_per_pixel];
[... 99 lines cut ...]
[InfoInfo] ] [Freenect2DeviceImpl] starting...[Freenect2DeviceImpl] starting...

Listening on port Listening on port 11111112......

[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [Freenect2DeviceImpl] started
device serial: 013286152847
device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [Freenect2DeviceImpl] started
device serial: 016305652847
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 768

xlz, 2017-06-15:
I think it's a hardware limitation. The bandwidth is not enough to handle a second Kinect.

rain3k, 2017-06-15:
i think so............ 
oh my god.............  hardware .... limitation...............................

closed 2017-10-13 by floe