OpenKinect/libfreenect2 #864: libfreenect2 ir verses ms kinect v2 sdk ir

timprepscius, 2017-06-13:
[image]
[image]

git log -1 --oneline
6236635 docs: Fix a typo

First of all, this project is great.  The code is clear.  Very little cruft.  I can modify it easily.  Kudos to all of you.

So, I am doing comparisons of libfreenect2 to the windows v2 sdk.
The libfreenect2 image is the first image.

For my purposes the windows v2 sdk IR image is more useful.  Do anyone have any thoughts on how they accomplish this normalization?    Before I start trying different strategies, I thought I would ask, perhaps someone has already tackled this or given it more thought than I have. (the last couple hours).

floe, 2017-06-14:
Very interesting. The deconvolution code which libfreenect2 is based on was extracted from the GPU shaders the MS SDK is using, but it's of course entirely possible that the algorithm has been updated since. We'd probably have to have another look at the shaders (/cc @christiankerl @xlz).

xlz, 2017-06-15:
I think the MS SDK normalizes the IR image by illumination correction template.

Start from here https://github.com/OpenKinect/libfreenect2/issues/144#issuecomment-132430933

In there, I was able to obtain something of similar nature from part of existing data though I still have no idea how exactly that data is intended to be used. In the same thread @christiankerl had something in his paper of similar nature, probably calibrated with his own methods, of which I had even less knowledge.