OpenKinect/libfreenect2 #866: libfreenect2 depth vs ms sdk depth

timprepscius, 2017-06-14:
git log -1 --oneline
6236635 docs: Fix a typo

So I'm looking very carefully at the depth image produced by the ms sdk and the libfreenect2 sdk.

MS:

[image]

lf2:

[image]

If you notice that cross object.  There is much more noise.

If I turn off the filters of the libfreenect2:

All filters:

[image]

Just Bilateral:

[image]

No Filters:

[image]

Unfortunately I can't do just edgeawarefilter.  Things don't run.

I am wonder two things:

Do we know how MS postprocesses?  Is it done via shaders?

I found a comment:
https://github.com/OpenKinect/libfreenect2/issues/144#issuecomment-117638164

Can you explain how you extracted the shaders?
Do you have the shaders for that version v2.0_1409?

I would like to check to see if they have changed.

xlz, 2017-06-15:
> Do we know how MS postprocesses? Is it done via shaders?

I think so.

> Can you explain how you extracted the shaders?

Shaders are probably included in the binary as strings. Use a reverse engineering/binary inspection tool I guess.

timprepscius, 2017-06-15:
Ok, so I've extracted the shaders from vvtechs.dll

gist of code to extract:
https://gist.github.com/timprepscius/e08fa2c9ae7e0cb232c3149ad6788fd8

and then I can run them through fxc.exe /dumpbin

gist of bat to do them all
https://gist.github.com/timprepscius/30fafe9002e82b93074c361ec1e4ea71

----

Here is the final result:
[vvtechs-shaders.zip](https://github.com/OpenKinect/libfreenect2/files/1077584/vvtechs-shaders.zip)

----

Ok, does anyone have a spare vvtechs.dll from 2016?? when the shaders were transmogrified into opengl for the current libfreenect2?

Or shader listings?  I would like to diff them, see what has changed.

floe, 2017-06-15:
Only @christiankerl might still have them somewhere. We never posted them publicly, as they are Microsoft-copyrighted code and we didn't want to give any reason for legal trouble.

floe, 2017-06-15:
But you're right, the old DLL should probably still float around. I might have it on my old VM, I'll check.

floe, 2017-06-15:
Found it (a really old one from ~2014): http://floe.butterbrot.org/external/vvtechs-2.0.1409.dll
Let me know when you've downloaded it, I'd like to remove the link again.

timprepscius, 2017-06-15:
Got it.

timprepscius, 2017-06-16:
Ok, the shaders are exactly the same.  But, as I've been reading the shaders, I notice there are more than a few variables not included by the lf2 opengl shaders.

Can anyone describe the process how the opengl shaders were created?  I do find the HLSL assembler very difficult to read through.  Is there a best tool to go about recreating the HLSL from the disassembled code?  Is there a best doc on the assembly instructions... The MS is a bit light on certain aspects.

My current plan is to use: https://github.com/baldurk/renderdoc
I am hoping I will be able to break on the shader IR and Depth, trace through, and see what is going on.

Any guidance based on past efforts would be great.  I will probably work on this next week though.

xlz, 2017-12-01:
I suppose this should merge into #865.

closed 2017-12-01 by xlz