OpenKinect/libfreenect2 #868: default packet pipeline choice incorrect (?)

svenevs, 2017-06-19:
Hello, I was reading through issues trying to figure out why my application is lagging so hard (protonect runs fine, clearly a me problem...).

I saw #656 linked at the bottom to the [performance wiki](https://github.com/OpenKinect/libfreenect2/wiki/Performance) expressly stating CUDA > OpenGL > OpenCL.

I think this should be reflected in [`createDefaultPacketPipeline()`](https://github.com/OpenKinect/libfreenect2/blob/6a63380ece335e07584ed55fcf7a0dd8b7fedf41/src/libfreenect2.cpp#L935-L943).  I think the order should probably be

1. Check for `LIBFREENECT2_WITH_CUDA_SUPPORT`
2. Check for `LIBFREENECT2_WITH_OPENCL_SUPPORT`
3. Check for `LIBFREENECT2_WITH_OPENGL_SUPPORT`
4. Fallback on cpu packet pipeline.

I didn't submit a PR because it's not exactly clear this is the right choice.

- If somebody installed using CUDA support, you will snag that as a priority.  Even if they also installed OpenCL support, CUDA will get used instead.
- For non-CUDA, try prefer OpenCL to OpenGL.
- Prefer OpenGL over cpu.

-----------------------------------------------------------------------

The pipelines are confusing to me though, I don't really understand their full purpose let alone what defaults should be.  At first glance the naming gives the impression that I could acquire device pointers (device = gpu), but [this doesn't seem to be the case](https://github.com/OpenKinect/libfreenect2/blob/d324e45f14f233e377012b369e234467efcf2839/src/cuda_depth_packet_processor.cu#L840-L859).  In the CUDA case, everything is being done on the default stream with a lot of `cudaDeviceSynchronize()` going on.  It might be nice allow users to keep the memory on the GPU with some fences or events (or whatever OpenCL does), but this level of exposure from the API would be far from easy to implement.........

Of course I was hoping this is why I was lagging (cuda-gl interop application), but the cpu packet pipeline has equivalent lag xD

xlz, 2017-06-20:
In terms of compatibility OpenGL > OpenCL > CUDA. So the default is OpenGL.

You want to driver support detected at runtime. I thought about it. It is too complicated (relative to its benefit).

> It might be nice allow users to keep the memory on the GPU

True. The plan was to amend `libfreenect2::Frame` to allow for passing GPU memory pointer, but the developers appear to have been all busy so far.

svenevs, 2017-07-13:
> In terms of compatibility OpenGL > OpenCL > CUDA. So the default is OpenGL.

Ok that makes sense, though I'd argue CUDA > OpenCL for compatibility [given NVIDIAs outright negligence](https://github.com/OpenKinect/libfreenect2/issues/804)) :wink:

> You want to driver support detected at runtime. I thought about it. It is too complicated (relative to its benefit).

Care to briefly elaborate on the complexity aspect?  Since everything related to available drivers has an associated compile time directive, my thought train leads me toward something like

```cpp
struct pipelines {
    static const bool cpu = true;

    #if defined(LIBFREENECT2_WITH_OPENGL_SUPPORT)
        static const bool opengl = true;
    #else
        static const bool opengl = false;
    #endif // LIBFREENECT2_WITH_OPENGL_SUPPORT

    #if defined(LIBFREENECT2_WITH_CUDA_SUPPORT)
        static const bool cuda = true;
    #else
        static const bool cuda = false;
    #endif // LIBFREENECT2_WITH_CUDA_SUPPORT

    #if defined(LIBFREENECT2_WITH_OPENCL_SUPPORT)
        static const bool opencl = true;
    #else
        static const bool opencl = false;
    #endif // LIBFREENECT2_WITH_OPENCL_SUPPORT
};
```

Maybe I'm missing the point though.  This would enable runtime queries of what is available, but you'd still ultimately need compile time conditionals floating around (e.g. to throw an exception if they try and create a cuda packet pipeline but didn't build with cuda).

Like you could also create methods that basically do the same thing

```cpp
bool hasOpenGLPacketPipeline() {
    #if defined(LIBFREENECT2_WITH_OPENGL_SUPPORT)
        return true;
    #else
        return false;
    #endif // LIBFREENECT2_WITH_OPENGL_SUPPORT
}
```

it's all ultimately up to the user of libfreenect2 to actually do something with it.  But now they can query based off their own preferences

```cpp
if(libfreenect2::hasCUDAPacketPipeline()) { ... }
else if(libfreenect2::hasOpenGLPacketPipeline()) { ... }
```

AKA runtime queries would be nice for users to have, and I don't think libfreenect2 would have to change at all -- just new methods available in a future release sort of thing.

I'm happy to write up actual code in a PR for you all if you want one of the above.  Sorry for the radio silence!

xlz, 2017-12-01:
Before I close this I'll explain a bit why it's complicated.

What you described is compile-time feature detection. The macros are detected by CMake. Runtime feature detection would be dlopen(some CUDA libraries) and call their interface to see if it works. dlopen(CUDA libraries) is very complicated.

closed 2017-12-01 by xlz

svenevs, 2017-12-01:
I see.  Runtime queries don't really seem like they are worth the effort.  And compiled-in-feature-detection does not guarantee (as you insinuate) that the pipeline will actually work.

In short, if a user identifies that they want a specific pipeline, or need to avoid a specific pipeline, they should create it explicitly **and** document it is the one being used.

If a developer identifies that they want to use a specific pipeline (e.g., the CUDA pipeline), if the user does not have that setup then the project won't even compile (`CudaPacketPipeline` will not be defined).  AKA the compile-time-queries don't really add any value.