OpenKinect/libfreenect2 #870: Need root access to use kinect

sdenenberg, 2017-06-20:
I've used libfreenect2 for quite some time now without any issues.  I just bought a new windows laptop, created a second partition with Ubuntu 16.04, and am trying to get my system up and running.  The issue I'm having is that I can't access the Kinects without using sudo before executing my program - this was never an issue before.  When I try to run Protonect, I get the same error:

[Error] [Freenect2Impl] failed to open Kinect v2: @2:3 LIBUSB_ERROR_ACCESS Access denied (insufficient permissions)

Running sudo ./Protonect solves that issue.  Any idea why I have insufficient permissions?  I'm definitely an administrator, and my user password is the sudo password.

Thanks,

Scott

lesterlo, 2017-06-20:
Did you copy the udev rules file?

"Set up udev rules for device access: sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/, then replug the Kinect."

sdenenberg, 2017-06-20:
Sigh, I knew it was going to be something obvious...  That sounds right, going to try it now.

sdenenberg, 2017-06-20:
Problem solved, thanks!

xlz, 2017-06-21:
Have a nice day.

closed 2017-06-21 by xlz