OpenKinect/libfreenect2 #873: jetson tx2  make error

koala99, 2017-07-04:
when i make file，
the error occured
[image]

MidoriYakumo, 2017-07-08:
interseting you can disable vaapi on nvidia boards cause you can use tegrajpeg

xlz, 2017-07-26:
True, build it with `cmake -DENABLE_VAAPI=OFF` or something.

closed 2017-07-26 by xlz

RBEGamer, 2017-08-09:
this works for eg : `cmake .. -DENABLE_VAAPI=OFF -DCMAKE_INSTALL_PREFIX=$HOME/freenect2`

leviresende, 2019-05-13:
Hey guys!
I'm using the JETSON TX 2 with the following configuration:
 L4T 28.2.1 [ JetPack 3.3 or 3.2.1 ]
 Board: t186ref
 Ubuntu 16.04.6 LTS
 Kernel Version: 4.4.38-tegra
 CUDA 9.0.252

And I'd like to connect it with a Kinect v2, however, I didn't have success. I've done everything that I read on the forums here on Github and on JetsonHacks, even tried to install as Kinect v1, but nothing. After all, I saw that it's not possible to connect a default board (TX2) with this device. Is it true?

By the way, which camera is it better to develop SLAM applications? ZED?

Thanks in advance.