OpenKinect/libfreenect2 #874: [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)

hyperrjas, 2017-07-06:
i have to restart ubuntu with command ./bin/Protonect

```
LIBUSB_DEBUG=3 build/bin/Protonect -nodepth
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: build/bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 000627663547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/gallium_drv_video.so
libva info: va_openDriver() returns -1
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 000627663547
device firmware: 2.3.3913.0
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
libusb: warning [handle_timeout] async cancel failed -4 errno=19
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
```
**uname -a**
`Linux ros 4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

**lspci | grep USB**

```
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 05)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 05)
02:00.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev ff)
```
kinect is connected in **NEC Corporation uPD720200 USB 3.0**

**free -h**

```
              total        used        free      shared  buff/cache   available
Mem:            15G        1.4G         13G         74M        992M         13G
Swap:           15G          0B         15G
```
**dmesg**

```
.
.
.
[   10.158934] r8169 0000:06:00.0 enp6s0: link up
[   10.158945] IPv6: ADDRCONF(NETDEV_CHANGE): enp6s0: link becomes ready
[   71.800214] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[   78.205122] usb 4-1.1: timeout: still 5 active urbs on EP #85
[   79.204911] usb 4-1.1: timeout: still 5 active urbs on EP #85
[   82.380193] xhci_hcd 0000:02:00.0: xHCI host not responding to stop endpoint command.
[   82.380196] xhci_hcd 0000:02:00.0: Assuming host is dying, halting host.
[   82.380252] xhci_hcd 0000:02:00.0: HC died; cleaning up
[   82.380390] usb 3-1: USB disconnect, device number 2
[   82.380558] usb 4-1: USB disconnect, device number 2
[   82.380560] usb 4-1.1: USB disconnect, device number 3
[   93.660227] xhci_hcd 0000:02:00.0: Stopped the command ring failed, maybe the host is dead
[   93.909999] xhci_hcd 0000:02:00.0: Host not halted after 16000 microseconds.
[   93.910001] xhci_hcd 0000:02:00.0: Abort command ring failed
[   93.910007] xhci_hcd 0000:02:00.0: HC died; cleaning up
[   93.910057] xhci_hcd 0000:02:00.0: Timeout while waiting for configure endpoint command
[   93.910064] usb 4-1.1: Not enough bandwidth for altsetting 0
```
Thank you

Abduoit, 2017-07-24:
Did you solve it, I am facing same issue, I have to restart ubuntu each time I run `./bin/Protonect`

xlz, 2017-07-26:
> Not enough bandwidth

That says all.

closed 2017-07-26 by xlz