OpenKinect/libfreenect2 #876: Multiple Kinect2 issues?

Xbbei, 2017-07-08:
Hello ,
 I want to use the libfreenect2 to connect multiple kinect2 and capture; My pc's operation system is Ubuntu 16.04 LTS, I replace the Protonect.cpp with the code: 
[Protonect.cpp.txt](https://github.com/OpenKinect/libfreenect2/files/1132953/Protonect.cpp.txt)

and I did the troubleshooting the `echo 64 > /sys/module/usbcore/parameters/usbfs_memory_mb`

But it will generate the error:
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 003570661547
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 003375361547
[Info] [Freenect2Impl] found 2 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to set configuration! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:6

So What's wrong with my code or my pc?

GoadingGoat, 2017-07-24:
Hi,
I'm also experiencing the same issues. I'm using debian stretch, with libusb 1.0.21
Not sure if @Xbbei noticed or not, but the libfreenect2 is trying to access the same camera twice (the camera that is found first). As to whether this is an issue with libfreenect2 or libusb, i am unsure....

I have tested this usuing my own code, that puts the frame listeners into separate threads, and by running two instances of Protonect, and both times it tried to access the same camera, both when the serial of the camera to access was specified, and when it was not.

lesterlo, 2017-08-15:
When I use Protonect to open two Kinects at the same time by using the serial number, 2 Protonect programs will sometimes open the same Kinect. But it doesn't happen when I using my own code by calling libfreenect2 API.

GoadingGoat, 2017-08-15:
@lesterlo I'm going to assume you are you using Windows from what you have described. I think the issue is with libusb...

On windows (with libusbk) using protonect without specifying device will open the same camera, but it is possible to open different cameras by specifying serials. Accessing the same camera on windows works in the sense that their are two device handles to one device (not sure if this is meant to be possible). You may notice severe issues with frame rate, probably depends if you use the blocking call for getting frames from the listener or not though.

Linux is a different story. Using protonect with no serials, will attempt to open the same (first) device. This generates the above error messages as described by @Xbbei. Using protonect and specifying serials also tries to open the same device, and generates the same messages. Opening the two handles to the same device on Linux does not work at all.

walterwang, 2017-09-13:
any luck in resolving this issue?

closed 2017-10-13 by floe