OpenKinect/libfreenect2 #877: How to install OpenCL correctly

smie14353078, 2017-07-09:
i am using ubuntu16.04+opencv3.2.0+Intel GPU
i install OpenCL with 
sudo apt-add-repository ppa:floe/beignet; sudo apt-get update; sudo apt-get install beignet-dev; sudo dpkg -i debs/ocl-icd*deb
it seems work properly however when i run 'clinfo' , get the results as follow
I think sth wrong with the OpenCL installing process but i dont know how to install it correctly

DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 1.2 beignet 1.1.1
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_spir cl_khr_icd
  Platform Extensions function suffix             Intel
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)

xlz, 2017-07-13:
The instructions said this:

> (Other) sudo apt-get install beignet-dev

So try it.

smie14353078, 2017-07-13:
thanks for your response xlz
actually that sentence is included into the long one 
and of cause i also have tried it independently
but the warning still exists

floe, 2017-10-13:
You don't need the old PPA anymore for Ubuntu 16.04. Just install the regular beignet package from the standard repos.

closed 2017-10-13 by floe