OpenKinect/libfreenect2 #878: CMake Error:LibUSB_LIBRARY

abcdvzz, 2017-07-13:
hey 
I tried installing the project on the Linux, when I get to this step:

cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2

there comes a problem:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
    linked by target "freenect2" in directory /home/leo/桌面/libfreenect2-master

how can i fix it?

svenevs, 2017-07-13:
It seems like you are running Linux based off the output above, so make sure you have both `libusb` and the development packages installed.  You have two options, inlined from [the install instructions for linux](https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux):

> - Install libusb. The version must be >= 1.0.20.
>     i. (Ubuntu 14.04 only) sudo dpkg -i debs/libusb*deb
>     ii. (Other) sudo apt-get install libusb-1.0-0-dev

If you go the `dpkg` route, make sure that you have followed the **Ubuntu 14.04 only** "Download upgrade deb files" into the `depends` folder.  

The high level issue here is that LibUSB is one of the few libraries that doesn't actually have cmake modules setup, which is why libfreenect2 is [vendoring its own `FindLibUSB.cmake`](https://github.com/OpenKinect/libfreenect2/blob/master/cmake_modules/FindLibUSB.cmake).  What's important to point out about how they've set things up (to make your life easier) is the lines at the very beginning:

```cmake
IF(PKG_CONFIG_FOUND)
 IF(DEPENDS_DIR) #Otherwise use System pkg-config path
 SET(ENV{PKG_CONFIG_PATH} "$ENV{PKG_CONFIG_PATH}:${DEPENDS_DIR}/libusb/lib/pkgconfig")
 ENDIF()
# ...
```

Basically, they make the assumption that if `DEPENDS_DIR` is set, always try and find the dependencies there first.  So if you used `apt-get` or `dnf` or whatever to install libusb, you need to do something like

```console
$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DDEPENDS_DIR=OFF
```

On linux, the other dependencies such as `glfw3` should be easy for you to get.  But note that `DEPENDS_DIR=OFF` affects many other portions of the build process (look in the files in the `cmake_modules` folder, they all generally follow the same approach).  So you'll want to make sure that you can find your dependencies

```console
$ pkg-config --cflags --libs libusb-1.0
```

An example of when you don't have the dependency findable by `pkg-config` would look like this:

```console
$ pkg-config --cflags --libs glfw3
Package glfw3 was not found in the pkg-config search path.
Perhaps you should add the directory containing `glfw3.pc'
to the PKG_CONFIG_PATH environment variable
No package 'glfw3' found
```

So just make sure that if you want to build the OpenGL stuff (you almost certainly do), you have glfw3 installed.  So on and so forth.

Basically, choose either to use the dependencies folder they setup for you (this is the easiest approach), or make sure that everything can be found using `pkg-config` from the command line.

Hope that helps in understanding these kinds of errors and how to track them down :smile:

abcdvzz, 2017-07-14:
Thank you, I can understand the process you referred to. But when I input this command: cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DDEPENDS_DIR=OFF. The same problem happens. And dependencies can be found when I input the command :pkg-config.........

svenevs, 2017-07-14:
I'm not sure what to tell you, you aren't giving very much information about what is going wrong.  Here are some suggestions:

1. When executing the new `cmake` command, it seems likely that the cache from `cmake` the first go around may be causing you grief.  I say this because the first time you ran things `DEPENDS_DIR` was not `OFF`, so you may get lucky if you

    ```console
    # remove the entire build tree
    $ cd /home/leo/桌面/libfreenect2-master

    # assuming you had just called it 'build'
    $ rm -rf build/
    $ mkdir build
    $ cd build
    $ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DDEPENDS_DIR=OFF
    ```

    Hopefully with a clean build tree things will "just work".

2. If not, you're gonna have to get dirty and tinker with the `cmake` build system.  More on that after
3. After you are able to get the LibUSB stuff working, I strongly encourage you use `ccmake` or the `cmake-gui` as opposed to setting everything you want from the command line.  For `ccmake`, the process looks something like this for a fresh clone:

    ```console
    $ sven:~/Desktop> git clone https://github.com/OpenKinect/libfreenect2.git
    Cloning into 'libfreenect2'...
    remote: Counting objects: 4734, done.
    remote: Total 4734 (delta 0), reused 0 (delta 0), pack-reused 4734
    Receiving objects: 100% (4734/4734), 3.01 MiB | 0 bytes/s, done.
    Resolving deltas: 100% (3038/3038), done.
    Checking connectivity... done.
    $ sven:~/Desktop> cd libfreenect2/
    $ sven:~/Desktop/libfreenect2> mkdir build
    $ sven:~/Desktop/libfreenect2> cd build/
    $ sven:~/Desktop/libfreenect2/build> cmake ..
    ```

    Even if it fails, what this did was create the "cache" for cmake.  So now I can run `ccmake` and potentially help `cmake` find the things it missed, as well as set options such as using `c++11`, enabling / disabling CUDA / OpenCL etc.  It's generally a lot easier to use `cmake-gui` or `ccmake` for these kinds of things than it is to try and get it all in by the command line.  So now when I run `ccmake` it looks like this

    ```console
    # now that we ran cmake it built the cache, `ccmake` is looking for the cache so '.'
    # as opposed to '..'
    $ sven:~/Desktop/libfreenect2/build> ccmake .
    ```

    Now that you have `ccmake` or the `cmake-gui` open, you'll be able to set whatever you need e.g. the `CMAKE_INSTALL_PREFIX`, etc.

-----------------------------------------------------------------------------

Using `ccmake` or `cmake-gui` will not help you in terms of getting it to find LibUSB.  So you'll need to experiment on your own and figure out why your dev setup isn't working.  If you just want to compile it and be done, it doesn't need to be robust, you just need to get it working right?  So here are the hacks you should employ:

1. Comment out the call to [`FIND_PACKAGE(LibUSB REQUIRED`](https://github.com/OpenKinect/libfreenect2/blob/70df5fa3aed0f007dcf98ba07d0cf55b2a7acd91/CMakeLists.txt#L93)
2. Manually set the variables [that script would have set](https://github.com/OpenKinect/libfreenect2/blob/70df5fa3aed0f007dcf98ba07d0cf55b2a7acd91/cmake_modules/FindLibUSB.cmake#L5-L8):

    ```cmake
    # FIND_PACKAGE(LibUSB REQUIRED)
    set(LibUSB_FOUND ON)
    # part a
    set(LibUSB_INCLUDE_DIR "...the path here...")
    # part b
    set(LibUSB_LIBRARIES "...the libs here...")
    ```

**Part A**: we need the directory.  Yours will look nothing like mine, but since on my machine I have

```console
$ pkg-config --cflags libusb-1.0
-I/opt/spack/opt/spack/linux-fedora23-x86_64/gcc-5.3.1/libusb-1.0.21-32c5olv7ym27lrztquwblwrmxnxuuli2/include/libusb-1.0
```

I'll remove the `-I` at the beginning and do

```cmake
# part a
set(LibUSB_INCLUDE_DIR "/opt/spack/opt/spack/linux-fedora23-x86_64/gcc-5.3.1/libusb-1.0.21-32c5olv7ym27lrztquwblwrmxnxuuli2/include/libusb-1.0")
```

**Part B**: pretty much the same thing, only with `pkg-config --libs`.

Good luck.

closed 2017-10-13 by floe

luchen828, 2020-06-08:
i solved problem of
 CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_LIBRARY
by setting parta and partb in FindLibUSB.cmake