OpenKinect/libfreenect2 #885: lib

Yvanou, 2017-07-21:
Hi everyone,

I work on MacOS X Yosemite (10.10.5).

When I do "cmake .." from the build directory of libfreenect2, I get this error message:

```
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
CMake Error at /Users/Yvanou/anaconda/share/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):
  A required package was not found

```

Yet, I correctly installed libusb using Homebrew.

Here is the complete log:

```
-- using tinythread as threading library
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
CMake Error at /Users/Yvanou/anaconda/share/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):
  A required package was not found
Call Stack (most recent call first):
  /Users/Yvanou/anaconda/share/cmake-3.8/Modules/FindPkgConfig.cmake:589 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/Users/Yvanou/Documents/libfreenect2/build/CMakeFiles/CMakeOutput.log".
Yvanou-MacBookPro2015:build Yvanou$ conda install pkg-config
Fetching package metadata ...............
Solving package specifications: .

# All requested packages already installed.
# packages in environment at /Users/Yvanou/anaconda:
#
pkg-config                0.29.1                        1    conda-forge
Yvanou-MacBookPro2015:build Yvanou$ conda search libfreenect
Fetching package metadata ...............
Yvanou-MacBookPro2015:build Yvanou$ pip search libfreenect
freenect2 (0.2.2)       - Python bindings for the libfreenect2 Kinect for Windows driver
pylibfreenect2 (0.1.2)  - A python interface for libfreenect2
Yvanou-MacBookPro2015:build Yvanou$ cmake ..
-- using tinythread as threading library
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
CMake Error at /Users/Yvanou/anaconda/share/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):
  A required package was not found
Call Stack (most recent call first):
  /Users/Yvanou/anaconda/share/cmake-3.8/Modules/FindPkgConfig.cmake:589 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/Users/Yvanou/Documents/libfreenect2/build/CMakeFiles/CMakeOutput.log".
```

Some idea in order to help me ?

Thanks a lot!

xlz, 2017-07-26:
libusb and libusb-1.0 are two different things.

closed 2017-07-26 by xlz

mwebb33, 2019-07-20:
Yeah uh, I am also having this issue. What is the resolution?

Funaizhang, 2019-07-22:
Same issue. Same ask. On macOS 10.12.
Tried solving this using #835, but the instruction is not clear and it still doesn't work.

cvirostek, 2020-04-06:
I installed libusb via Macports instead of Homebrew and it fixed this issue for me.